Jyers/Marlin #2: [BUG] Manual leveling Z axis

Sennevds, 2021-01-28:
Hi really nice fork you've made. I think I found a bug (or I'm not using the feature right) when I tried to manual level (I don't have a bltouch) the nozzle will move to the point but the Z axis won't go down. 
So first I Auto Home nozzle goes to bottom left until it hits the endstop then I select bottom right nozzle goes up goes to the point but doesn't go down at the point. (same for the other points)

Schnabulation, 2021-01-28:
I have the same fork installed and I just went and checked my printer: For me it works perfectly. First I auto-home, then I go to manual leveling option and select bottem right, printhead moves to bottom right and Z-axis drops down until the Z-axis switch is hit.

BUT: I have a BL Touch configured and a Z-offset set.

Sennevds, 2021-01-28:
Hmm I've compiled the firmware myself and changed 2 things the size of the bed and I'm using the 4.2.7 board so I've changed that but normally that shouldn't be an issue I think.

Jyers, 2021-01-29:
I'll try flashing the firmware without the bl touch and test to see if I can recreate your issue, but the only thing I can imagine causing such a bug would be some kind of configuration issue.

Sennevds, 2021-01-29:
Here's my configuration.h and configuration_adv.h: https://gist.github.com/Sennevds/3fe10fb27d43945c84af9f1d5e1f2386

Sennevds, 2021-02-01:
The new manual bed level works perfect so I'm going to close this issue

commit 118b734f (2021-03-21): Manual Probing Preheat Fix #2
commit d3ac1373 (2021-05-12): Conditional Fix #2
closed 2021-02-01 by Sennevds