Jyers/Marlin #4: [BUG] Print status screen switches to "Host Print" when printing locally from display.

Schnabulation, 2021-01-30:
### Bug Description

When OctoPrint is connected to the printer and a print is started directly from the display the status screen switches to "Host print" instead of the normal, locally started "print from SD card" screen. This renders the "pause" and "cancel" buttons functionless.

### Configuration Files

Please find my config-files in my fork: https://github.com/Schnabulation/Marlin/tree/Ender_3_V2_Rewrite/Marlin

### Steps to Reproduce

1. Connect OctoPrint to the printer and configure the M75 to M77 commands in OctoPrint.
2. Insert a SD card with GCODE and start a print directly from the printer.
3. After the temperatures are up and the print starts watch the display switch to "Host print" screen.

**Expected behavior:**

The screen should keep the normal local "Print from SD card" status screen showing the filename instead of "Host print".

**Actual behavior:**

The screen switches to "Host print" rendering the buttons useless.

#### Additional Information

We already discussed this issue and you seemed to have fixed it. However it now arises again in actual prints.

Schnabulation, 2021-01-30:
I just checked again and the switch from the default status screen to the "host print" screen happens when OctoPrint switches its status from "Starting print from SD" to "Printing from SD".

Schnabulation, 2021-02-01:
I think this has been fixed in the latest commit. Did you address this issue at all?

Jyers, 2021-02-01:
Yes, I added a check to make sure the card is not printing since octoprint sends an M77 command on start even if it is not handling the print. It should be fixed now.

closed 2021-02-01 by Jyers