Jyers/Marlin #5: Live adjust causes flicker and unexpected behaviour.

luketot, 2021-02-01:
1st,
Screen flickers when enabling Live Adjust from the Z Offset menu. Looks like its trying to show an alert.
After playing around some more, I see the alert while the Z axis moves to 0.0.

2nd,
After more playing around, it seems to be misbehaving a little: when I enable Live Adjust, the head moves to 0.0, and not what the z-offset is set to.
Set offset to 5mm,
Enable Live Adjust, (head should move up to 5mm, it doesn't)
Set offset to 5.01mm, (to see if it will move to the proper height with a new command, it doesn't. Instead it just moves up the 0.01mm difference)
Set offset to 0mm, (head lowers by 5mm, crashing into the bed)

I went to test what happens if I adjust z-offset when the z height is already up, such as after a print, but it seems that live adjust turns off as soon as you exit that menu. Not sure if that's intentional. Should live adjust influence changing z-offset during a print?

Jyers, 2021-02-01:
Thanks for the report. The flicker is inescapable as live adjust runs G28 O when its enabled to make sure its homed. This requires a homing popup, but if its already been homed it will only happen for a split second. 

The offset issue is interesting. Do you have a bed probe? It is set to go to Z 0, but if an offset has been applied Z 0 refers to the offset position, not the actual zero, so I don't know how it could be returning to zero,

Jyers, 2021-02-01:
Found the problem. The latest commit resets the z position to zero after moving it with the z offset menu. This seems to have fixed the issue.

closed 2021-02-01 by Jyers