Jyers/Marlin #6: [BUG] Part of the popup stays on screen

Schnabulation, 2021-02-01:
### Bug Description

After the last change (commit #5d20df03cd62c18d873313dc1669403dd0e6bdd2) some part of the grey popups (for homing, leveling, etc.) stays on the screen even after the popup disappeared.

Before homing:
[image]

During homing:
[image]

After homing:
[image]

See the grey bar at the bottom.

Jyers, 2021-02-01:
Thanks for the catch, this is due to the addition of extra space so that M117 messages can be displayed in the menus. I will update the popup size to not stretch down quite so far

Jyers, 2021-02-01:
Latest commit moves the bottom of the popups up a bit to rectify this issue.

closed 2021-02-01 by Jyers