Jyers/Marlin #11: [BUG] Status screen remains, info screen never reappears using M600 in g-code

robby1367, 2021-02-05:
When using M600  in g-code, the filament retracts and the popup appears.
After swapping filament and pressing confirm, the filament loads and printing continues.
The popup remains on the screen and when pressing confirm, the screen returns to the prepare menu.
The info (print status) screen does not reappear until the print is finished and the progress bar shows the value at the point of the filament change.
[configs.zip](https://github.com/Jyers/Marlin/files/5929303/configs.zip)

Schnabulation, 2021-02-05:
I can confirm this: also happens on my printer. However I don't mind it: if you have another M600 in the same GCODE file it will open the popup again just fine.

Jyers, 2021-02-05:
This is definitely something I will fix, I will probably just have to add a printing flag so the popup can tell if it should go to the prepare menu or the tune menu

Legsmaniac, 2021-02-09:
Glad I found this, was about to report the same bug.  
Yes, same thing happens to me.  
I can press the button and I'm back on the home screen with no print progress bar or print time.
And I like the print time so I'll await this fix.  :-)

Thanks for a great Marlin mod.  Found this quite by chance looking for Filament Runout information.  👍

Jyers, 2021-02-09:
Sorry for the delay, this should be fixed now. I can't test right now, but I'll close this as soon as the fix is confirmed.

Legsmaniac, 2021-02-09:
It's half fixed.

After my filament ejected, there was no "Continue" message on the screen as before?
So I was unsure what stage it was at.

When I loaded the filament, it seemed to only take in less than half the length it's supposed to whereas before, it took in what I set it to eject - 400mm, so I had to push it through myself the rest of the way.

I seemed to have to press the button about three times to get it started again.
However, all the while, it stayed on the print information screen so yes, I do have the progress bar and print times.
Then I noticed the hotend temp was wrong.  I started at 210C - it was now 180C
I had to adjust the temp quickly to set it back to 210.

robby1367, 2021-02-10:
Now the popup does not appear, it does stay on the tune screen. . After changing filament, pressing play continues and appears to be functioning correctly.

Legsmaniac, in the ADVANCED_PAUSE_FEATURE define, there are the defines for unload and load lengths and feed rates that need to be set for your configuration. There is also a timeout for the nozzle heater that could be changed to allow you more time to change filament.

Legsmaniac, 2021-02-10:
I thought I had already set up ADVANCED_PAUSE_FEATURE settings as it was working before the fix.
Somehow it's got reset.  My Bad!  

So, thanks to robby1367 for nudging me on that and apologies to Jyers, the fault is mine entirely.
Although I don't understand why the bed temp set itself to zero and the extruder to 180.
I have gone through the configuration files with a fine tooth comb tonight/early this morning and given that it's now very late, I'll upload the new firmware later in the morning and test it again.  Knackered now, time for bed.
I'll report events later - stand by.........

Legsmaniac, 2021-02-10:
OK, woke up for some pills and a coffee so did a quick test.
Confirm it's working as expected.
Thanks again to robby for the heads up to what to check and correct.  👍
Many thanks to Jyers for the fork and the quick fix.  👍

Jyers, 2021-02-10:
Glad to know that did the trick!

closed 2021-02-10 by Jyers