Jyers/Marlin #13: [FR] Suggestion: Only park X with NOZZLE_PARK_FEATURE with Ender 3 V2

nullstalgia, 2021-02-05:
### Description

Since you have enabled the Filament Change by default, I wanted to throw this in. 

I think only parking X would make sense, as that's the only Axis that has the hot-end attached.

Also (and please inform me if this is just wrong), but I feel that sending Y to 0 for no reason during a print could result in the bed possibly being misaligned with the print. But, I feel this would happen most if filament changes/pauses happened multiple times during a print, not with just one pause.

Also, when pausing a print to say, add a magnet, the bed moving all the way to the back makes it a lot harder.

t3chguy, 2021-02-05:
That's unrelated to this fork, that's based on your Advanced Park configuration in base Marlin.
The likelihood of misalignment is the same as skipped steps during a print in general.

Jyers, 2021-02-05:
As @t3chguy said, this is a base marlin feature and is not something that I should really modify if I ever hope to get this into the main marlin branch. I agree with your reasoning, but its just not a feasible edit

closed 2021-02-05 by Jyers

Schnabulation, 2021-02-05:
I'd like to add here that in configuration.h is already a setting: #define NOZZLE_PARK_X_ONLY

If this is defined it will only park X - so exactly what you want.

nullstalgia, 2021-02-05:
Fair enough, just wanted to see how it would be responded too. Thanks for
your time!