Jyers/Marlin #14: [FR] Manual Mesh leveling - "Previous Point"

tinooo, 2021-02-06:
Doing the manual leveling, it would be amazuing, having another menu entry for "previous". Unfortunately in a 5x5 levling it happens quite often (to me, and guess to others too), that you press the scrollwhellt accidentally/to quick when adjusting a new point. That results in pressing the "next" button, leaving your current point at the default offset of 0.2mm.

Now if you go through 5x5 levling and this happens somewhere in the middle/end, it's quite cumbersome, either doing it manually by RS232, or going through all points again.

As per my primitive understanding, this might be an easy fix, using G29 I .. J ... command and set it to the previous point.

Jyers, 2021-02-09:
I have not tried using I and J with manual leveling. There is a chance that these options may be solely for directly editing a mesh value rather than moving to that point to re-measure it, but I will test to see if this could be a potential option for going to a previous point.

mosheas, 2021-02-10:
After I do a Manual Mesh leveling calibration and make a save for all point, I go back for check test again Mesh leveling and the value goes back to 0.2mm, the last calibration doesn't keeping on memory , how do I fix it?

techtosterone, 2021-02-12:
> After I do a Manual Mesh leveling calibration and make a save for all point, I go back for check test again Mesh leveling and the value goes back to 0.2mm, the last calibration doesn't keeping on memory , how do I fix it?

Uncomment "RESTORE_LEVELING_AFTER_G28" - so mesh is not lost after homing.

watch this at 11-min mark
https://www.youtube.com/watch?v=38ugKOKD7YM

mosheas, 2021-02-12:
While I do a manual mesh leveling calibration there is only an option for the next calibration , it is also possible to make an option to go back eveling calibration?

Benetos, 2021-02-16:
If you switched to unified bed leveling you could have slots. S1 S2 S3. Bilinear might support it.

olskar, 2021-02-25:
> Uncomment "RESTORE_LEVELING_AFTER_G28" - so mesh is not lost after homing.
>
> watch this at 11-min mark
> https://www.youtube.com/watch?v=38ugKOKD7YM

Is RESTORE_LEVELING_AFTER_G28 enabled in the precompiled firmware @Jyers ? :)

Jyers, 2021-02-25:
> Is RESTORE_LEVELING_AFTER_G28 enabled in the precompiled firmware @Jyers ? :)

Yes :)

gokhanselen, 2021-02-28:
Dear JyersI installed the software you sent me. I am very pleased. Thank you. I will be very happy if you answer me with 4 questions.
1.ender 3 v2 makes a sound when I open it.
2.When testing with the bltouch, the bltouch is pressed on the tray in the left front corner.
3.I bought a Filament Sensor. Here the filaments are broken too much. What software should I install when I insert the filament sensor?

I want to support you but I am not working because of covid19

It will be my first task when I start working

Thanks for your help
Gokhan Selen
(I am sorry if there is a spelling mistake, I am writing with a Turkish-English translation.)

Jyers, 2021-02-28:
1. The sound is most likely from the steppers engaging, but I'm not sure. What kind of sound is it?
2. Do you mean the BL Touch is only going to the front left corner and then failing, or is something else happening?
3. For installing a filament sensor, I would recommend compiling your own firmware so you can configure the sensor properly. There are plenty of tutorials online.

Hopefully all that makes sense when translated back to Turkish :)

gokhanselen, 2021-02-28:
Jyers Thank you very much for your answer. -The sound comes from the motors making a harsh sound to 'full' while being ready  -Bltouch gets idle when it comes to the left front and gives an error it should start from the right a little but I don't know how to get it  Jyers I have no knowledge (0) on my computer and software issues I'm trying to learn how to use it I use the software you have compiled I want your help because I can't compile Where can I find software with bltouch and filament sensor for Ender3 v2 -4.2.7 board or do you have the software? Thank you for your help. You look like a really good person because you help Gokhan Selen (google translate translates well)

bsturn, 2021-03-01:
Hey @Jyers , quick question. I have the manual mesh leveling installed and when I turn my machine off/on it loses the mesh it generated. Is this by design or just the way the machine operates? So far I’m absolutely loving the extensible UI and the precomplied firmware has worked perfectly, just some minor text issues on my display but that’s likely from the TH3D LCD software. Many thanks to you!

Andrewqld, 2021-03-03:
With the Manual mesh leveling, why is the firmware set in such a way that it loses the mesh it generated, any chance it could be set to save this when printer is turned off?
And yes, I know it can be done in the precompiled firmware but I am stupid and cannot seem to be able to load the firmware without visual studio showing 56 faults and refusing to compile it.
By the way, I love the firmware, thanks so much.

Jyers, 2021-03-03:
That is very strange, it should not be losing the mesh on restart. The mesh is saved directly to eeprom right after leveling, so I don't see how it could be getting lost... Are you sure it is really losing the mesh? If you run M420 from a host after a restart does it show anything?

bsturn, 2021-03-04:
It seems to have figured itself out? Not quite sure. Either that or my bed became madly unlevel while wiping it down. Which is probably the case, although I didn't apply THAT much force. Not quite sure but it's fixed now. Keep up the great work man! Absolutely loving it so far.

Andrewqld, 2021-03-04:
I haven't tried it in a while, I'll give it another go, thanks for the reply.

Jyers, 2021-03-25:
Unfortunately I don't think this exact functionality can be achieved. However @kjeldahl's new PR allows the previous mesh value to be restored which should make a miss-click significantly less detrimental.

closed 2021-03-25 by Jyers

gokhanselen, 2021-04-17:
Merhaba Jyers menü yüklemek istiyorum ama dosya bulamadım yardımcı olurmusun Windows 10 için Posta ile gönderildi