Jyers/Marlin #21: [BUG] Layer shift after M600 or M0

arthurgeek, 2021-02-14:
First of all, thank you very much for this firmware. It's great.

I don't know if I should open a bug to Marlin directly or your fork, it's late night here and I haven't checked if it also happens in the official firmware. Will flash official firmware tomorrow and check. For now, I'll open this here.

I want to change filament mid print. I used Cura `Filament Change`  script (which uses `M600`) and `Pause at height` (`M0`) . Both options shift layers when print returns. `M600` seems to shift both X and Y and `M0` shifts Z (unsure if it also shifts X and Y, as shifting Z made a mess of my test print).

I'm building the firmware using your configs (from `Configuration\ Files/Jyers\'s\ Config`) and my own numbers for `NOZZLE_TO_PROBE_OFFSET`. No other change.

Jyers, 2021-02-14:
Interesting issue. I haven't heard of anyone with similar problems, but please do test to see if the issue occurs on the main branch as if it is a problem with my firmware I will definitely look into it.

nullstalgia, 2021-02-15:
Actually, I've been having this issue as well! I guess its not just me then. I thought it was my filament being weird, but I also have had bad adhesion to the old layer with the new filament pretty consistently with this branch. Now, it could just be more apparent due to the functionality being easier to use, but I dunno.

Schnabulation, 2021-02-15:
I have not tested M600 after the latest build, but I have tested it extensively some commits back and never had this issue.

Is it possible that it is hardware-related? Maybe the belts are not tentioned enough? Did you leave PAUSE_PARK_NO_STEPPER_TIMEOUT defined, so the stepper motors don't lose the position?

nullstalgia, 2021-02-15:
PAUSE_PARK_NO_STEPPER_TIMEOUT is still defined. And I have the nice twang with my belts. (Can't speak for Arthur)

Schnabulation, 2021-02-15:
Wierd... Basically the display just sends the M600 command to Marlin - the rest is handled by Marlin and I doubt that Jyers touched that code. I'll test with the newest commit...

Schnabulation, 2021-02-15:
I **can not** confirm this issue on my printer - tested as follow with the newest firmware (pulled, compiled and flashed as of right now):

1. Sliced a xyzCube in Cura with M600 between layer 19 and 20.
2. Printed on my printer and watched for the change filament popup
3. Unloaded and loaded the same filament and confirmed the prompt.
4. Also tested a manually induced filament change with the "Tune" menu during print.
5. ...same procedure.

No layer shifts at all - cube printed perfectly.

nullstalgia, 2021-02-15:
I will copy your procedure and see if I can reproduce it with my machine. (Won't be for a little while, unfortunately.)

Jyers, 2021-02-17:
As no more updates have been provided and it seems like a main marlin issue, I'm closing this for now.

closed 2021-02-17 by Jyers

arthurgeek, 2021-02-17:
@Jyers yes, it seems to be a Marlin issue.

I had a chance to try printing this again yesterday and after making a small change, I was able to succeed. I noticed that during M600 with X and Y as 0 the nozzle would crash the X stopper really hard. Changing it to something like 20,20 made it work for me. Was able to print without an issue. This was tested with this firmware, I did not flashed stock Marlin on it.

Schnabulation, 2021-02-17:
If I may ask something about this: in my configuration_adv.h I have `PARK_HEAD_ON_PAUSE` **undefined** but the head still parks when I send a normal M600 command. Any idea why?

Jyers, 2021-02-17:
I believe M600 always parks the head, park head on pause is just telling it to also do so on a normal pause event

Schnabulation, 2021-02-17:
Ah, this makes sense. Thank you!

tommymonk, 2021-03-17:
I'm experiencing this issue, I had an M600 that was meant to allow me to start printing a new filament at `4.2` and when the head returned to resume printing it was `4.19`, I had to reduce Z by `0.17` as quick as I could

```
G0 F300 X88.972 Y116.444 Z4.2
G0 F9000 X88.906 Y116.383
G0 X90.1 Y114.69
G0 X89.342 Y113.915
G0 X80.158 Y68.909
G0 X79.791 Y65.884
G0 X79.651 Y62.624
G0 X79.804 Y59.107
G0 X80.215 Y55.932
G0 X80.934 Y52.6
G0 X81.934 Y49.333
G0 X82.895 Y46.885
G0 X84.169 Y44.222
G0 X88.686 Y46.598
;TIME_ELAPSED:6824.725354
M600 X30.00 Y30.00 ; Generated by FilamentChange plugin
;LAYER:20
```

Jyers, 2021-03-17:
This is definitely a gcode issue of some kind, check after the filament change to see what height it tries to go to. Also, did it say the correct height on the display or did it display the actual height it went to (ie the wrong height)

tommymonk, 2021-03-17:
I don't see any zheight change immediately after the `M600` I assumed moving back to previous position was internal to it.

The display on the printer (I'm printing from octoprint) did show `4.19` instead of the `4.2` I expected

Here is the gcode https://gist.github.com/tommymonk/3d240e309da6c78d63fb5da14257fb81

Jyers, 2021-03-17:
Interesting, I don't know what could cause that. Maybe try another print with a filament change and see if you get the same occurrence.

tommymonk, 2021-03-17:
I can reproduce with a 1cm square solid cube

Schnabulation, 2021-03-17:
You do realize that 4.19 is literally 0.01mm lower than 4.2? That is like a 20th of a layer height - that shouldn‘t be a problem at all.

tommymonk, 2021-03-17:
Really ?

My layer change needed to start at layer 21 and I was printing .2mm layer height, so when it stops for filament change and the "z" value on bottom right of screen says "4.0" I assume it's reading in millimetres 20 x 0.2 = 4mm

Schnabulation, 2021-03-18:
> value on bottom right of screen says "4.0" I assume it's reading in millimetres 20 x 0.2 = 4mm

Yes, this is correct. The coordinates are listed in mm. But the difference between 4.19 and 4.2 is 0.01mm and not 0.17mm as you mentioned. Or is there a typo and you ment 4.**0**2? That would be a problem indeed. But a difference of 0.01mm is so abysmally small that it should not induce a layer shift.

tommymonk, 2021-03-18:
Yes I messed up the calculation, it seems to always resume about the correct number but seems to sometimes be a couple of layers too high, I've checked that it isn't an issue with fitment of my z hardware e.g. backlash / slop

I'm now following this https://github.com/MarlinFirmware/Marlin/issues/19067