Jyers/Marlin #22: [BUG] M0 pause locks display

Spirko, 2021-02-14:
### Bug Description

The M0 command now prompts the user for a button press (awesome!), but when I do it, the screen reverts to a short vertical orange bar in the upper-left corner, no icons, and won't allow interaction.  The three status lines do seem to be still updating.

### Configuration Files

[Bugreport Configuration.zip](https://github.com/Jyers/Marlin/files/5976995/Bugreport.Configuration.zip)

Not too many changes.  Here are my changed lines:
`//#define USE_PROBE_FOR_Z_HOMING`
` #define NOZZLE_TO_PROBE_OFFSET { -45, -25, 0.1 }`
`   #define GRID_MAX_POINTS_X 5`
`//#define Z_SAFE_HOMING`

### Steps to Reproduce

1. Power-on Ender 3 V2
2. Using Pronterface, send `M0 "Press the button"`
3. Press the knob button.

**Expected behavior:**

Display returns to previous screen, in this case the main menu.

**Actual behavior:**

Display shows no icons and a small vertical orange bar in the upper-left

#### Additional Information

This is with branch upstream/Ender_3_V2_Extensible_UI, commit "Slight Changes and Updates" dated 2/13/2021 6:52:09 PM.

Photo:  https://photos.app.goo.gl/C3PHeqAtASddkQ9g9

Jyers, 2021-02-14:
I am aware of this issue, but thanks for posting it! The extensible UI port is still in its early stages so I wouldn't recommend using it until I get most of the kinks worked out. I will post an update on reddit once I feel it is ready for community testing. If you do want to build it occasionally and report any bugs you find your welcome to, but in general its not the most stable thing to run for everyday use.

Spirko, 2021-02-14:
I'm probably also not supposed to issue a pause command while the main menu is displayed.

Spirko, 2021-02-14:
If it helps in tracking down a solution, this only happens from the main menu.  The orange box is the selection indicator for the itemized menus.

Jyers, 2021-02-14:
Thanks for the help, I've got it fixed. I just hadn't fully implemented the confirm control yet for extui based calls. I just need to test it then I'll push the commit.

Spirko, 2021-02-14:
Fixed in "Fixed User Confirm Prompts".  Great work!

closed 2021-02-14 by Spirko