Jyers/Marlin #23: [BUG] Fix warnings and still allow configured preheat labels

Spirko, 2021-02-14:
In the ExtensibleUI branch, `Marlin/src/lcd/extui/lib/creality_dwin/creality_dwin.cpp`, function `CrealityDWINClass::Get_Menu_Title()`, I was getting warnings about using `strcat()`.  Since these string constants, they can be concatenated by juxtaposition.  Example:

    case Preheat1:
      return (char*)(PREHEAT_1_LABEL " Settings");

Jyers, 2021-02-14:
Wow! I had fixed that issue by removing the settings bit all together but that is a much better solution! I've been programming is C++ a long time so I love it when I get to learn something new!

Spirko, 2021-02-14:
That screen makes sense with just PREHEAT_1_LABEL, so your solution works, too.

Jyers, 2021-02-14:
I implemented your solution, thanks again for the new knowledge about constants!

closed 2021-02-14 by Jyers