Jyers/Marlin #44: [BUG] M73 showes wierd percentages in new branch

Schnabulation, 2021-02-16:
### Bug Description

In the new Ender_3_V2_Extensible_UI branch the M73 command gives some wierd percentages on the screen when printing from OctoPrint:

[image]

### Configuration Files

Find my configuration files in my fork: https://github.com/Schnabulation/Marlin/tree/Ender_3_V2_Extensible_UI/Marlin

The most importand configs would be these:

`#define LCD_SET_PROGRESS_MANUALLY`
`#define USE_M73_REMAINING_TIME`

I am using [M73 Progress Plugin](https://plugins.octoprint.org/plugins/m73progress/) and [DisplayLayerProgress Plugin](https://plugins.octoprint.org/plugins/DisplayLayerProgress/) in OctoPrint.

### Steps to Reproduce

Just print normally through OctoPrint. You could also check by manually sending a "M73 P25" through the terminal.

Jyers, 2021-02-16:
I can confirm this issue. Based on my testing it seems to be multiplying the percent give by 100 and the strange number you see is an overflow error.

clintiepoo, 2021-02-17:
Same issue here - if you need another test case.

Jyers, 2021-02-17:
This is fixed in the build #1.1

closed 2021-02-17 by Jyers