Jyers/Marlin #62: Either X-Axis and Y-Axis move very slow occasionally

Sherman-c, 2021-02-17:
HI Jyers

Thank you for your hard work, I like bed level menu, that saved me a lot of time to do bed leveling, I found one issue on your firmware which was not showing on stock marlin bugfix-2.0.x.  
If I moved Z-axis over 150+, and move either x-axis or y-axis or both, they will move very slow, almost 1 seconds per step. Also when I did the estep calibration, the e step motor move very fast compared with marlin stock.

kind regards
Sherman

Jyers, 2021-02-17:
The extruder issue is fixed in the latest build. As far as the strange movement issue, I will have to test it. I'll update this once I've checked it out.

Sherman-c, 2021-02-17:
Hi Jyers/Marlin

Thanks for the update,

Kind regards
Sherman

Jyers, 2021-02-26:
I was unable to recreate this issue, my process was:
* Raising the Z axis to 175
* Moving X and Y
* Repeated at Z 200

I did not notice any form of slowness. If you are still able to recreate the problem on the latest release feel free to post an update here and I can reopen the issue.

closed 2021-02-26 by Jyers