Jyers/Marlin #70: Progress Bar remains at 0%

DerT94, 2021-02-17:
Hey,

When I print over octoprint, the printer says "Host Print" but the progress bar stays at 0%.
I use the OctoPrint plugin "DisplayLayerProgres" and your pre compiled firmware "E3V2-Default-NoProbe.bin"

BR
T

Deletedice, 2021-02-17:
In addition to this, the remaining time is also at 0. It seems as if the display is not receiving the code estimate

chevyman142000, 2021-02-17:
Interesting. I am not having this issue. I'm using the m73 progress and displaylayerprogress plugins. Have you also updated the GCODE scripts in Octoprint with the commands? https://github.com/Jyers/Marlin/blob/Ender_3_V2_Rewrite/Images/Octoprint%20Settings.JPG

Jyers, 2021-02-17:
Are you sure "DisplayLayerProgres" sends M73 commands? As far as I can tell it looks as though it only sends M117 messages

Jyers, 2021-02-17:
@Deletedice Are you sure you have an octoprint plugin that is set up to send M73 commands?

Deletedice, 2021-02-17:
> @Deletedice Are you sure you have an octoprint plugin that is set up to send M73 commands?

I realised that i didnt have the GCODE scripts set as it was not in the files that I had downloaded. Ive set the commands as @chevyman142000 has showed. 
Printing rn so cant say if it worked or not just yet however i feel as if this is the solution

Deletedice, 2021-02-18:
@Jyers Updated code as shown on picture above with M73 command, Still showing 0% and 00:00 remaining.

Jyers, 2021-02-18:
The picture had nothing to do with M73 commands. You need a plugin for octoprint that sends M73 commands to the printer. such as "M73 Progress" https://plugins.octoprint.org/plugins/m73progress/

t3chguy, 2021-02-18:
Or use PrusaSlicer which leaves `M73`s in the GCODE itself

Schnabulation, 2021-02-18:
Can confirm: with M73 Progress it works flawlessly... Tested today.

Deletedice, 2021-02-18:
Apologies! I missed the part about the plugin. Ive installed and now is working correctly. Thank you and sorry again

closed 2021-02-19 by Jyers

DerT94, 2021-02-19:
Hey,
I also missed the part aput the M73 Progress Plugin.
Now i installed it and the prograss bar is active but nor correct.

After the print start (OctoPrint 1%)
[image]

After 1 Minute (OctoPrint 3%)
[image]

After 2 Minutes (OctoPrint 6%)
[image]
[image]

Schnabulation, 2021-02-19:
You need to install the latest version of the firmware - there was a bug creating these wrong values.

It's fixed in the latest version - I can confirm this.

DerT94, 2021-02-19:
Is the pre compiled file not the latest version?
What I have to configure to get the same configuration as in the precompiled file "E3V2-Default-NoProbe.bin"