Jyers/Marlin #78: [FR] multilanguage support

ArteInverso, 2021-02-18:
I am studying the multilanguage implementation on this screen with the previous versions and it made it difficult to carry out this task, but with the migration to Extensible UI I see this implementation more possible

# Description

Language implementation es, fr, it ...... on the ender 3 v2 screen

#This is a snippet of the tests I started

 // Allow language selection from menu at run-time (otherwise use LCD_LANGUAGE)
#define LCD_LANGUAGE_1 en
#define LCD_LANGUAGE_2 fr
#define LCD_LANGUAGE_3 de
#define LCD_LANGUAGE_4 es
#define LCD_LANGUAGE_5 it

If it is of your interest or someone is already working on this, it is my intention that we can do this as a team and be able to bring a better experience to users around the world.

Jyers, 2021-02-19:
I would be very interested in implementing such a thing if people are interested in collaboration. I think the best approach would be for me to set up the framework of code for it to function, and then leave it to anyone interested in adding their language to create a set of string constants for the system.

Schnabulation, 2021-02-19:
How many strings are to translate? I could do German if it‘s not too many...

gokhanselen, 2021-02-19:
Türkce olmaz mı

ArteInverso, 2021-02-19:
> I would be very interested in implementing such a thing if people are interested in collaboration. I think the best approach would be for me to set up the framework of code for it to function, and [...]

I would like to be able to help carry out this work, I have already seen some things on how to implement it but it would be good to organize steps to follow, I will be helping with the translation into Spanish and other primary languages

gokhanselen, 2021-02-20:
Eğer yardım edebilecek bir şeyim olursa yardımcı olmak isterim 
Ender3 v2 e anakart bozuldu 4.2.7 kart aldım bltouch da var 
Hangi yazılımı önerirsiniz 
Saygılarımla
Gökhan SELEN 

iPhone’umdan gönderildi

bulgaro80, 2021-04-05:
I'm available for Italian, just let me know what to translate

esloch10, 2021-04-06:
I can help for french and arabic language .. if it's support

Tuncay-Ayhan, 2021-04-06:
I can translate to Dutch if you want

robilabet, 2021-04-10:
I can translate to Portuguese (Pt-Br) to help

Mineplayerminer, 2021-04-14:
I can do Slovak language.

mlask, 2021-04-16:
I can do Polish (pl-pl) translations.

Also, it's worth mentioning that there's [font dump](https://github.com/enigmaquip/T5UIC1_Font) of DWIN T5UIC1 available. With access to the right tools it's possible to create own font for that screen.

JP-rg, 2021-04-17:
I can get Spanish done, I normally live there but the Pandemic

N3tYaroz3, 2021-04-22:
I am available to translate into Italian

Jyers, 2021-04-22:
@N3tYaroz3 Nothing yet currently, I need to implement the language system and then I'll update this with what's needed.

N3tYaroz3, 2021-04-23:
> @ N3tYaroz3 Al momento non c'è ancora niente, devo implementare il sistema linguistico e poi lo aggiornerò con ciò che è necessario.

In the meantime, thank you for what you are doing!
Please take me into consideration if you need a translator for the Italian language, I would be happy to help you and I will gladly do it! 😊

Tuncay-Ayhan, 2021-04-28:
> @N3tYaroz3 Nothing yet currently, I need to implement the language system and then I'll update this with what's needed.

Maybe look into https://app.lokalise.com/ . I've seen people use it with the [Home-Assistant project](https://github.com/home-assistant/core) .

Joulss, 2021-05-07:
Available for french if needed

Qubi-B, 2021-05-13:
If u don't have anyone I can help with polish

agalakhov, 2021-05-13:
I can help with Russian. I also could find a helper for Japanese.

imax10000, 2021-06-10:
I can provide German translations, but I will need some instructions on creating "a set of string constants for the system" as @Jyers suggested

Jyers, 2021-06-10:
@imax10000, I may still try to use the marlin language system rather that writing my own, but if need be there will be a template that I will distribute for users to fill out with their contributed language.

agalakhov, 2021-06-11:
Which fonts does the DWIN module support? I don't have any docs.

mlask, 2021-06-11:
> Which fonts does the DWIN module support? I don't have any docs.

Look [here](https://github.com/enigmaquip/T5UIC1_Font) - standard font contains basic latin characters in few sizes and GB2312 (chinese, japanese, greek, cyrillic) character set in one size only.

TuriionCZ, 2021-06-24:
Hi guys, can translate to czech language ;)

araziforum, 2021-06-25:
I can help for translate Turkish

jonasl03, 2021-08-01:
Count me in for danish if this gets implemented

akechagias, 2021-11-08:
I can do the Greek version, if it is supported!

dhupee, 2021-11-24:
I can help with the Indonesian/Helping Malay once it implemented

dante-35, 2021-12-08:
For Turkish tranlation i can help too.