Jyers/Marlin #82: Extrudeur (Menu Prepare->Move) not reset 0.0 after Print or STOP Print

tititopher68, 2021-02-19:
Hello,
First of all, well done for this firmware! He is almost perfect. Almost because there is something that deserves a correction. The extruder. In the Prepare / Move / Extruder menu ... The Extruder parameter is not reset to 0.0 after a Print or after a STOP Print. Also It is difficult to simply make for example a recalibration of the extruder (by measuring 120mm of wire, then extruding 100 mm and then by measuring the piece of remaining wire to calculate the parameter of the extruder ....) without having to turn off the printer ....
During Print, if you do a STOP Print, the extruder indicates for example 300.6
If I display 100 and validate, it retracts 200.6 mm! To have an extrusion of 100mm, I have to validate 400.6.
But if the extruder showed 460.2 for example, I cannot display 560.2, because limited to 500 .... So there is a problem ...

Schnabulation, 2021-02-19:
I believe this is how stock-Marlin behaves as well. There is a GCODE command (G92 E0 I believe) to reset the extruder but otherwise not much. You could however create a menu item that just sends this command to the board...

tititopher68, 2021-02-19:
Just tested with the Marlin Stock and no issue.
Just this issue with this last firmware (Ender_3_V2_Extensible_UI)

Schnabulation, 2021-02-19:
Can you make a step by step description on how to test this?

tititopher68, 2021-02-19:
With your firmware:

1 / On the first Photo (photo1), the value displayed for the extruder parameter (just after the end of  my last print) is 52.6 with your firmware.

2 / Now i want to check the calibration of my extruder.
I mark the filament at 120 mm (photo2).

3 / I display 100 on the screen and I validate (I want to extrude 100 mm of filament - photo3).

4 / Only 100-52.6 mm or 48.4 mm were extruded .... (photo4).

See : [https://e.pcloud.link/publink/show?code=VZekMkZ4YWSqWIVJI4OrqnnfNiW6h5NBca7](url)

With a Marlin-bugfix-2.0.x firmware (the one from 2 days ago):

1 / The displayed parameter (under the same conditions) is 0.0

2 / Now I want to check the calibration of my extruder.
I mark the filament at 120 mm (as in photo2).

3 / I display 100 on the screen and validate.

4/100 mm of filament have been extruded ...

Jyers, 2021-02-20:
I am definitely going to change how the extruder item functions in the move menu. Just for reference, does the extruder value get reset to 0 every time you click on the item to change it, or is it only when you go into the menu?

tititopher68, 2021-02-21:
Hello, 

I modified creality_dwin.cpp in /src/lcd/extui/lib/creality_dwin/   like this 👍 

        case MOVE_E:
          if (draw) {
            Draw_Menu_Item(row, ICON_Extruder, (char*)"Extruder");
            Draw_Float(current_position.e, item);
          }
          else {
            if (thermalManager.temp_hotend[0].celsius < EXTRUDE_MINTEMP) {
              Popup_Handler(ETemp);
            }
            else {
              gcode.process_subcommands_now_P(PSTR("G92 E0"));
              Modify_Value(current_position.e, -1000, 1000, 1);
            }
          }
          break;

And it's now OK...

tititopher68, 2021-02-21:
Hello,

I modified creality_dwin.cpp in /src/lcd/extui/lib/creality_dwin/ like this 👍

    case MOVE_E:
      if (draw) {
        Draw_Menu_Item(row, ICON_Extruder, (char*)"Extruder");
        Draw_Float(current_position.e, item);
      }
      else {
        if (thermalManager.temp_hotend[0].celsius < EXTRUDE_MINTEMP) {
          Popup_Handler(ETemp);
        }
        else {
          gcode.process_subcommands_now_P(PSTR("G92 E0"));
          Modify_Value(current_position.e, -1000, 1000, 10);
        }
      }
      break;
And it's now OK...

Jyers, 2021-02-21:
I implemented a similar thing to go into the next release. It just uses some marlin core functions instead of gcode, and also resets the value when you enter the menu

Jyers, 2021-02-23:
This has been implemented in the latest release

closed 2021-02-23 by Jyers