Jyers/Marlin #98: Problem with resume print does not start part cooling again

mimes123, 2021-02-22:
I use the pre compiled version on the E3V2 3x3 bin file.
and just tested the pause print option. 

When pressing resume print after a pause it did not start the part cooling again.

Jyers, 2021-02-22:
Thanks for the report. I figured the gcode would restart that, but I guess I was wrong. I'll get this fixed for the next release.

Schnabulation, 2021-02-22:
Hang on, I will first give this a test.

Did you start/stop from the display or OctoPrint?

Jyers, 2021-02-22:
@Schnabulation Be aware that because they are using a precompiled bin, its the default pause and not an M125 pause and park.

Schnabulation, 2021-02-22:
Is that M0 or M25? Do you know what pause the display will send when hitting „Pause“ on the screen?

Jyers, 2021-02-23:
It uses M25 with some extras that I added in to allow for accurate reheating

Jyers, 2021-02-23:
Fan speed is resumed after pause in the latest release

closed 2021-02-23 by Jyers