Jyers/Marlin #118: Bar panel display Percentage does not match time

mosheas, 2021-02-24:
Percentage
[image]
 bar panel display does not match time remaining times & Elapsed Time, please see attached photos.

Jyers, 2021-02-24:
Is that purely an SD print, or is there a host connected? Because I didn't think the printer displayed time remaining in the status bar like that without a host sending M117 commands.

feelfree69, 2021-02-24:
Currently printing a larger part, 10h17m printing time according to Cura slicer.

It's at layer 7 out of 325, in .gcode file its at line #23000 out of 303000.
Progress bar shows 7% (which matches to 23000/303000).
Time display: Elapsed 1:39, Remaining 2:07

From what calculation comes the "Remaining" time currently?

I can't remember having such large discrepances with stock firmware, but I'm not 100% sure.
Simple estimation would be: 7% took 1:39, then 93% will take 21:55.
Hmm. still a factor of 2 away from the reality.... and in absolute numbers even worse than today....

mosheas, 2021-02-24:
Is it possible to calculate the percentage bar from a figure of M117 'Time
left"?
this calculation would be very accurate.

Schnabulation, 2021-02-24:
Can you guys please say if you printed from SD card or from OctoPrint?

I‘m using a about a week old version of this firmware with OctoPrint and always have perfectly accurate times on the display. I do use PrintTimeGenius though...

mosheas, 2021-02-24:
From SD card

feelfree69, 2021-02-24:
Also printing from SD card. I just found the option in CURA slicer to include M117 for time remaining and M73 for Percentage , that should do the job.

mosheas, 2021-02-25:
yes, Although I work with SD with M117 + M73 by Cura it's happening.

tititopher68, 2021-02-25:
hello,

In my opinion, there is nothing abnormal. I am on the latest version v1.1.1. The version doesn't matter. In addition it depends on the slicer. With Cura and even according to Cura versions, the estimated time is not exact.
You also need to know this. The system calculates the time remaining based on the layer being printed. If you are on a slow part, the time will increase. If you are on a faster part, this time will decrease.
Towards the very end of printing, the remaining time + the elapsed time will agree with the estimated total time expected for printing. 
It works like this for me. At the end, around 95%, the times are quite consistent.

Jyers, 2021-02-25:
Alright, I think I now understand that the slicer is what is sending these M117 messages reading 'Time Left' In which case, yes that will be far more accurate than the linear approximation the printer uses. If you would like more accurate times/progress on an sd print, you would need a slicer that can send M73 progress and time remining commands

closed 2021-02-25 by Jyers

mosheas, 2021-02-25:
OK, thank you. 👍