Jyers/Marlin #120: No print progress on display

olskar, 2021-02-24:
First of all, thank you for a firmware that makes my Ender 3 v2 useable!

### Description

I have read somewhere that when printing from octoprint, this firmware is able to show the progress on the display. I am not getting that.

#### Additional Information
This is my progress in octoprint
[image]

t3chguy, 2021-02-24:
Make sure you have https://plugins.octoprint.org/plugins/m73progress/ installed.

olskar, 2021-02-24:
> Make sure you have https://plugins.octoprint.org/plugins/m73progress/ installed.

Thank you! I do not have that.

Is there a difference between m73 and this for example (m117)?

https://plugins.octoprint.org/plugins/displayprogress/

feelfree69, 2021-02-24:
just learned it myself: m117 shows remainng time; m73 shows percentage.
Also Cura is able to include these commands, so also for SD card prints this should go fine.

t3chguy, 2021-02-24:
M73 tells the printer a percentage
M117 tells it a string to display.

The former can be used for things like the progress bar
The latter can do things like tell you layer number

olskar, 2021-02-24:
Thank you all, installing m73 and closing :)

closed 2021-02-24 by olskar