Jyers/Marlin #121: Display gets freezing after long pause

mosheas, 2021-02-24:
Hi... 
After long pause and when park nozzle is time out, i  pushing reprint, than Display gets freezing with 
a message "Reheat Finished Press To Continue" , even though the printer continues to print. only turning it off and on again it's fixes the phenomenon.
[image]

Schnabulation, 2021-02-25:
I had this happen to me, but pressing Continue again made the screen disappear and it returned to the status screen - that didn‘t work for you?

mosheas, 2021-02-25:
No it's not work on me, it's happening only after park nozzle is time out,
thank you

Jyers, 2021-02-26:
Are you on the latest build? Because this exact issue should be fixed in v1.1.1

mosheas, 2021-02-26:
Hi
Yes, of course I use with the latest version, this phenomenon has been
fixed in the past in the context of" filament sensor Runout".
Thank you.

Jyers, 2021-02-26:
I was unable to recreate this issue. My process:
* Start SD print
* Pause the print
* Wait for heater to timeout
* Click to reheat
* Press Resume
* Wait for reheat finished popup
* Click popup to confirm

After clicking, the screen returned to the print screen as expected. If you continue to experience this issue and would like to share a video or images of exactly how it is occurring feel free to update this thread and I will reopen the issue.

closed 2021-02-26 by Jyers