Jyers/Marlin #123: Display not updating after print from SD

jjfs127, 2021-02-24:
**Description:**

I usually have a small .gcode file (petg-load.gcode in the image) to load filament for my dual gear extruder from the SD card. Once loading is complete, I confirm the print and then proceed to print using Cura/Octoprint. I notice that uppermost of the display still retains the load gcode filename and progress bar remains at 100%. 

https://imgur.com/a/H792QeC

**Steps to Reproduce:**

1. Print from SD card and confirm print
2. Send print via Octoprint
3. Monitor display

**Expected behavior:**

Update top portion of display

**Actual behavior:** [What actually happens]

Upper portion of display does not update, please see image.

Deletedice, 2021-02-25:
So the print doesnt update from when you print via Octoprint? If thats the case have a look here https://github.com/Jyers/Marlin/issues/70 - You need to have M73 commands sent to the printer to update the file progress

jjfs127, 2021-02-25:
I just noticed that Octoprint started up in safe mode before printing, so this in fact could be attributed to this...

closed 2021-02-25 by jjfs127