Jyers/Marlin #124: Mesh Bed Leveling issue on version 1.1.0

Sherman-c, 2021-02-25:
Hi Jyers

I found 2 issues on manual mesh bed leveling after installed E3V2-ManualMesh-5x5-V4.bin, version 1.1.0, on my printer, 
When i doing mesh bed leveling manually, under Level icon, i found
1. It displays -0.00 on z-axis when I move nozzle down below 0.00, I would expect -0.01 after 0.00, not -0.00
2. occasionally the value on z-axis not changing 

This image is showing -0.00 on z-axis when i press move down
[image]
 

This video shows I pressed twice on 0.29 to reach 0.30,  0.28->0.29->0.29->0.30
https://user-images.githubusercontent.com/68487284/109097215-03956980-7773-11eb-88fe-9f0a12eabe68.mp4

Regards
Sherman

rorywquin, 2021-02-25:
following

t3chguy, 2021-02-25:
There's a subscribe/unsubscribe button in the right panel, you don't need to spam the thread.

[image]

rorywquin, 2021-02-25:
No intention  to spam the thread, but thanks for your pointer.

Jyers, 2021-02-26:
The strange behavior of needing two clicks for the value to change is due to rounding discrepancies with the position values and will not cause any issues with the final mesh. The issue of negative positions is certainly interesting and I will test to see if I can recreate it.

Jyers, 2021-02-26:
I was unable to recreate the negative position issue. It may just be due to the rounding of the values as I stated before. None the less, this should not impact the quality of your mesh.

closed 2021-02-26 by Jyers