Jyers/Marlin #125: Display failed to update after printing completed - Version 1.1.1

Sherman-c, 2021-02-25:
With version 1.1.1, after printing job completed, summited by Octoprint, the Display still waiting for confirmation.

[image]
[image]

Schnabulation, 2021-02-25:
This is the normal behaviour: when a print job is finished the „confirm“ button appears and needs to be pressed. Same when printing from SD card.

Jyers, 2021-02-26:
> This is the normal behaviour: when a print job is finished the „confirm“ button appears and needs to be pressed. Same when printing from SD card.

This is correct, after any print, host or sd, a user confirmation is required to return to the main menu. This however does not impact the starting of a new print from octoprint, as the screen will still update back to the print screen if a new print is started.

closed 2021-02-26 by Jyers