Jyers/Marlin #127: Filename not updating after printing from SD card

jjfs127, 2021-02-25:
**Description:**

I usually have a small .gcode file (in this case petg-load.gcode in the image) to load filament for my dual gear extruder from the SD card. Once loading is complete, I confirm the print and then proceed to print using Cura/Octoprint. I notice that uppermost of the display still retains the load gcode filename and doesn't update with the new file printed from Octoprint. 

https://imgur.com/a/H792QeC

**Steps to Reproduce:**

1. Print from SD card and confirm print
2. Send print via Octoprint
3. Monitor display

**Expected behavior:**

Update top portion of display (filename)

**Actual behavior:**

Upper portion of display does not update and retains previous filename, please see image.

Jyers, 2021-02-26:
When you say "confirm print" do you mean that it finished the sd job and you click confirm, sending you back to the main menu? Or are you starting the octoprint job while the print screen is still displayed?

jjfs127, 2021-02-26:
Hmm... I would think that I confirmed the previous print. But maybe you're right, perhaps I didn't make it all the way back?

Jyers, 2021-02-26:
It should redraw the print screen when a new print is called, so unless maybe the boolean for printing from SD didn't get updated I don't know what could cause that. If it happens again and you have any more info about it feel free to post an update here, otherwise I will try to recreate it when I have some time.

Jyers, 2021-02-26:
The fix for this should be live on github now. If you are using a precompiled firmware it will be in the next release.

closed 2021-02-26 by Jyers