Jyers/Marlin #147: [BUG] Flow rate setting - no effect

tinooo, 2021-02-28:
When adjusting the flow rate from the tune menu, nothing changes in the print.

I've verified it by setting the flow to 80% in the tune menu and printed a test cube. Then I resliced that cube and lowered the flow in the slicer to 80% and reprinted (this time with 100 in the tune menu).

The prints are clearly different.

Using Release 1.1.1
Not sure if I'm doing something wrong, but seems like a bug to me

t3chguy, 2021-02-28:
It definitely sets Marlin's flow rate correctly, I confirmed it has the same behaviour as setting it via GCODE from OctoPrint - it just uses M221 https://marlinfw.org/docs/gcode/M221.html

Schnabulation, 2021-02-28:
Does that mean f.e. when setting flow-rate to 150% and sending a `G1 E100` it would extrude 150mm of filament?

t3chguy, 2021-02-28:
That's what https://marlinfw.org/docs/gcode/M221.html seems to suggest but is not very clear on.

Jyers, 2021-03-02:
I have tested modifying the flowrate mid print with confirmed change, but I have not tested setting the flowrate and then attempting a print. I will have to give this a try. Just as a note, was this with a host print or with an SD print?

tinooo, 2021-03-02:
Printed from a SD card, didn't test from host!

xantrk, 2021-03-03:
If my understanding is right, I think slicer multiplies the flow rate with E values in GCODE itself, so it is not visible on screen. 

The flow rate in screen, I believe, uses M221, which will internally multiply E commands with M221 commands value. 

I was trying to get rid of some blobs in a small print, in my experience also setting the flow through screen has a bigger visible effect. I am not sure if all other variables such as leveling of bed is exactly the same, but %97 flow in cura with %97 flow in screen has turned out much better than %94 Cura - %100 screen.

tome9111991, 2021-04-04:
I have pinted 2 times same file.
1. With 100% flow set in lcd
2. With 75% flow set on lcd

I see no differencea

MariaKonopnicka, 2021-04-04:
I also see no difference of in flow when changing it during printing.

Jyers, 2021-04-06:
I think I finally fixed this with the latest commit. Update this if anyone still experiences the issue.

closed 2021-04-06 by Jyers