Jyers/Marlin #149: letters remain on both sides of the bottom bar,

mosheas, 2021-03-01:
When there is a file name is too long, the letters at the edges of the bar are not deleted, letters remain on both sides of the bottom bar, see photo. thank you. 
[image]

bsturn, 2021-03-01:
I am also having this. Although not sure if it's the LCD firmware I'm using or if it's the printer firmware

feelfree69, 2021-03-04:
Also having this with stock LCD firmware.

Jyers, 2021-03-05:
I would be very interested to know exactly what was displayed, as it is coded such that nothing should ever reach the edges.

mosheas, 2021-03-05:
The phenomenon only appears when the name of the printed file is very long, in my case about 40 letters, in short file names it does not happen.

feelfree69, 2021-03-05:
> I would be very interested to know exactly what was displayed, as it is coded such that nothing should ever reach the edges.

Just print a file from SD card with a very loooooooooong filename - longer than the space of the display so that the filename is scrolling.

Jyers, 2021-03-05:
Yes, I just implemented filename scrolling yesterday. How is it working? Are there characters getting stuck at the edges?

feelfree69, 2021-03-05:
I#m using your release 1.1.1.
Currently printing with ling filename:
1. At the very top, the filename is *not* scrolling. Whole width of display is being used.
2. Above the Tune/Pause/Stop Icons, the filename *is* scrolling. Leftovers at left and right edges.
3. Below the Tune/Pause/Stop Icons, currently no filename is being displayed. But Leftovers at left and right edges.

Jyers, 2021-03-05:
Ah, I see. Top filename scrolling isn't in the last release. I will have to test to see if the issue still exists in the current source version

Jyers, 2021-03-08:
This issue should now be fixed in the latest source and will be part of the next release.

closed 2021-03-08 by Jyers