Jyers/Marlin #152: Manual leveling does not disable ABL

wob8, 2021-03-01:
### Description
Noob alarm: I am quite new to this, build firmware from main Marlin branch a couple of times. Read about this firmware on reddit and used the smith3d before that.

TLDR: Manual bed leveling does not disable bed leveling (M420 S0) which is expected as any action could have enabled it.
-G29/'Level' from LCD
-Print just ran with G29 or M420 S1
-G28/'Auto Home from LCD' with the firmware option to enable ABL after homing

What happened:
After running Leveling from the ABL and looking at the report in octoprint terminal I noticed it was out of wack significantly.
I decided to use the LCD to run manual leveling option. It went to the 4 corners with perfect z.
So I tried again and issued an M420 after homing when in the menu where you can choose to go to the corners.
It reported bed leveling was enabled.
After setting M420 S0 and re-homing manual leveling went fine. The ABL afterwards was also fine.

I am using E3V2-BLTouch-5x5-HS-V4.bin v1.1.1 from 22nd of Feb

### Steps to Reproduce

1. Startup printer
2. <Optional> Raise one corner of the bed a bit so it'll be noticed
3. Run Leveling from LCD
3. Wait till done and save to EEPROM
4. From menu start manual bed leveling option
5. Send M420 command and confirm ABL is enabled.
6. Go to the 4 corners, watch ABL being applied (can see z-rod moving)

**Expected behavior:**
Bed leveling being disabled

**Actual behavior:**
Bed leveling enabled

Schnabulation, 2021-03-01:
I noticed that as well. However - if I want to manually level the bed I just reset the EEPROM from the menu and do it afterwards. I also auto-level before every print so I don‘t need to have the mesh in EEPROM.

Also btw: the Z-offset menu is also relying on the stored mesh - if there is none the offset is wrong.

Jyers, 2021-03-02:
Interesting, I never considered this issue. I'll add a command to disable the mesh when you enter the manual leveling menu.

Jyers, 2021-03-05:
This should now be fixed

closed 2021-03-05 by Jyers

Schnabulation, 2021-03-05:
Can you explain how you fixed it? Does manual leveling disable the mesh and re-enable it afterwards or is it not dependant on the mesh at all?

Jyers, 2021-03-05:
It runs and M420 S0 when entering the manual leveling menu and runs M420 S1 when you exit the menu.

Schnabulation, 2021-03-05:
PERFECT! Thx