Jyers/Marlin #153: frozen display bug & file name disappeared

mosheas, 2021-03-01:
In the latest update, the frozen display bug returned after Filament Runout, and the file name also disappeared from the top bar, please see attached photos, thanks. 
[image]
[image]

mosheas, 2021-03-04:
[image]

Jyers, 2021-03-05:
I must have introduced this when trying to fix some host print issue, Will work on a fix.

Jyers, 2021-03-05:
This should be fixed

closed 2021-03-05 by Jyers