Jyers/Marlin #159: FIL_RUNOUT_STATE HIGH/LOW always detects no filament

mccolister1017, 2021-03-02:
Start a print and it will instantly detect no filament what can i do ??? i got this fialment sensor  
[image]

mccolister1017, 2021-03-02:
Start a print and it will instantly detect no filament present

mosheas, 2021-03-02:
For me it works well, set low in firmware, and do a cross between the wires +volt and a signal in plug 
that connects to the board.
[image]

mccolister1017, 2021-03-02:
> For me it works well, set low in firmware, and do a cross between the wires +volt and a signal in plug
> that connects to the board.
> [image]

thank you i will try it

Schnabulation, 2021-03-04:
I just installed a filament runout sensor myself and only uncommented `#define FILAMENT_RUNOUT_SENSOR` in Configuration.h - worked like a charm.

mccolister1017, 2021-03-04:
> I just installed a filament runout sensor myself and only uncommented `#define FILAMENT_RUNOUT_SENSOR` in Configuration.h - worked like a charm.

thank you i will try because the first solution was ok but only for a time .... my runout sensor was laggy with it

Jyers, 2021-03-07:
Closing this due to inactivity. Feel free to post an update if the issue persists and I will gladly reopen the issue.

closed 2021-03-07 by Jyers