Jyers/Marlin #166: [BUG] Compile fails when you uncomment #define FILAMENT_RUNOUT_DISTANCE_MM 25

Legsmaniac, 2021-03-03:
### Description

When you uncomment the line `#define FILAMENT_RUNOUT_DISTANCE_MM 25` in `Confiruration.h` then attempt to compile using Platformio, it fails with the error ....

```
Marlin\src\lcd\extui\lib\creality_dwin\creality_dwin.cpp: In member function 'void CrealityDWINClass::Menu_Item_Handler(uint8_t, uint8_t, bool)':
Marlin\src\lcd\extui\lib\creality_dwin\creality_dwin.cpp:1980:31: error: 'FilamentMonitor {aka class TFilamentMonitor<RunoutResponseDebounced, FilamentSensorSwitch>}' has no member named 'runout_distance'
             Draw_Float(runout.runout_distance(), row, false, 10);
                               ^~~~~~~~~~~~~~~
Marlin\src\lcd\extui\lib\creality_dwin\creality_dwin.cpp:1983:33: error: 'FilamentMonitor {aka class TFilamentMonitor<RunoutResponseDebounced, FilamentSensorSwitch>}' has no member named 'runout_distance'
             Modify_Value(runout.runout_distance(), 0, 999, 10);
                                 ^~~~~~~~~~~~~~~
*** [.pio\build\STM32F103RET6_creality\src\src\lcd\extui\lib\creality_dwin\creality_dwin.cpp.o] Error 1
============================================= [FAILED] Took 45.45 seconds =============================================

Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  FAILED    00:00:45.454
======================================== 1 failed, 0 succeeded in 00:00:45.454 ========================================
```

### Steps to Reproduce

1. Download the latest current copy of Ender_3_V2_Extensible_UI
2. Copy both configuration files from the folder Configuration Files > E3V2 Templates > ManualMesh-3x3
3. Paste into the Marlin folder overwriting the originals
4. Comment out the `#define FILAMENT_RUNOUT_DISTANCE_MM 25` in `Confiruration.h` and save
5. Compile with Platformio (I use command line)

**Expected behavior:** [What you expect to happen]

I expect the compilation to succeed as it always has using previous versions.  I have a copy of the pervious download and tested this compiled successfully. 

**Actual behavior:** [What actually happens]

It starts to compile then fails with the message as above.

#### Additional Information

I assume this is a bug?  Something has been changed since the previous version?  I have never used `#define FILAMENT_RUNOUT_DISTANCE_MM 25` before and really don't need it.  In order to get this latest version to compile, I had to leave it uncommented and set it to 1mm then I was able to compile successfully.  
Was the change intentional?  Perhaps setting to 0 (zero) mm would be the same as commenting it out? (I never actually tried using zero)  Seems a strange change if so.

Jyers, 2021-03-03:
It would seem I forgot to add conditionals for when filament runout distance is disabled, I'll get this fixed and committed. Thanks for the catch!

Jyers, 2021-03-03:
Just pushed the fix :)

closed 2021-03-03 by Jyers