Jyers/Marlin #169: OctoPrint -> Printer Connectivity Issue

jamesbrindle, 2021-03-03:
Hey,

Great work BTW.

I've attempted to a few different releases:

* https://github.com/Jyers/Marlin/releases/tag/v1.1.1
-- E3V2-BLTouch-3x3-HS-V4.bin
-- E3V2-BLTouch-5x5-HS-V4.bin

* https://github.com/Jyers/Marlin/releases/tag/v1.1.0
-- E3V2-BLTouch-3x3-HS-V4.bin
-- E3V2-BLTouch-5x5-HS-V4.bin

And I seem to be having the issue where OctPrint won't connect? It connects sometimes after numerous Pi reboots and printer reboots, but not actually been able to print anything. When I do the OctoPrint disconnects again and I again have trouble connecting?

Has anyone else had this issue?

I flashed the older firmware back and the connectivity issue went away, but I'd really like to use your firmware for the LCD progress sync...

I'm using Ender 3 V2, 4.22 board, Pi 4.

Also, might I ask, is '5x5-HS' (the 'HS' - High speed part) is that bed levelling speed?

Any help would be greatly appreciated.

darco415, 2021-03-04:
Thats a tough because i  have connection issues too but I dont know If I can blame this FW...I too have an RPI and sometimes thinks the usb connector is going bad on the Pi or even the Ender..Ive used 3 different anker usb cables. Most of the time it runs fine but lately alot of re-compiling and flashing...Flashed a compiled UBL today and had issues with connection but most likely my fault...was able to finally probe table but then lost connection...reverted back...

Lumics, 2021-03-04:
Have you tried to set a fixed Baudrate instead of Auto?

Schnabulation, 2021-03-04:
I am also using OctoPrint with this firmware (granted, I compiled it myself, but that shouldn't change anything) and my connection is stable. I even have a 15m active USB-extension and a non-powered USB-hub in between my OctoPrint server and my printer - works flawlessly. Serialport and Baudrate both set to auto.

jamesbrindle, 2021-03-04:
> Have you tried to set a fixed Baudrate instead of Auto?

I was using an auto baudrate, I will try using a fixed one and see if it makes a difference.

jamesbrindle, 2021-03-04:
Still no luck I'm afraid... Oh well

Schnabulation, 2021-03-04:
Have you tried with your own compile? Or only with the pre-compiled ones?

jamesbrindle, 2021-03-04:
With both yes. I'm currently looking through the .h files for anything amiss or different from my previous marlin firmware, but starting to feel like giving up, anything  'connectivity-wise' appears to be the same.

Schnabulation, 2021-03-05:
Have you tried another cable and another port on the Pi?

I honestly don‘t know what it might be - I have a very stable OctoPrint connection with this firmware.

Sennevds, 2021-03-05:
I also had some problems with the connection but nothing that had to do with the fw (I think) If you have a disconnect you should check dmesg if there's something like following error: `ftdi_sio ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32` you have the same prooblem as me. The only thing that helped was changin RPI

https://github.com/raspberrypi/linux/issues/2406

jamesbrindle, 2021-03-05:
> I also had some problems with the connection but nothing that had to do with the fw (I think) If you have a disconnect you should check dmesg if there's something like following error: `ftdi_sio ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32` you have the same prooblem as me. The only thing that helped was changin RPI
> 
> [raspberrypi/linux#2406](https://github.com/raspberrypi/linux/issues/2406)

Hey. 

I appreciate that, I am using an RPI 4... I believe there's some random issues with RPI 4 and OctoPrint... But... The stock firmware works, and this firmware works (which I'm currently using): https://marlin.crc.id.au/firmware/

What RPI are you using?

There must be something that's different somewhere, but heaven knows I can't find the culprit. Tried different USB cables, different ports... It's a mystery.

jamesbrindle, 2021-03-05:
So I figured out what he issue was!

Decided to try another Pi as suggested (used a Pi 3 this time), so flashed OctoPi and low and behold it worked!

Still not convinced, I contemplated the fact that I'd just installed a fresh image of OctoPi and it maybe OctoPrint that was the issue, so I reflashed the Pi 4 and it worked!  So the issue was with OctoPrint, not the firmware...

After doing some digging and reinstalling plugins, I discovered it was specifically the **MeatPack** plugin that caused the connectivity issue. After uninstalling that, the printer connects - Seemingly quicker than it did before.

t3chguy, 2021-03-05:
> After doing some digging and reinstalling plugins, I discovered it was specifically the MeatPack plugin that caused the connectivity issue. After uninstalling that, the printer connects - Seemingly quicker than it did before.

Did you have the MEATPACK option enabled in the firmware Configuration?

jamesbrindle, 2021-03-05:
> > After doing some digging and reinstalling plugins, I discovered it was specifically the MeatPack plugin that caused the connectivity issue. After uninstalling that, the printer connects - Seemingly quicker than it did before.
> 
> Did you have the MEATPACK option enabled in the firmware Configuration?

Evidently not... I wasn't aware there was an option and wasn't looking for it. Myers did also state in a reddit that it's not enabled by default.

darco415, 2021-03-05:
Right, you have to manually enter #define MEATPACK in adv.h. I have meatpack enabled but my occasional disconnects were before adding meatpack. I still think its the usb connector on the PI...

sympotonic, 2021-03-06:
You need a better USB Cable from the pi to the printer. known issue for octopi with all firmwares out there.

jamesbrindle, 2021-03-06:
The issue was MeatPack, but thanks.

closed 2021-03-06 by jamesbrindle