Jyers/Marlin #174: [BUG] When M600 gets triggered via filament runout sensor the popup does not disappear afterwards

Schnabulation, 2021-03-04:
### Description

When `FILAMENT_RUNOUT_SENSOR` is defined and a filament runout is detected the printer correctly executes a `M600` command to move the head away and eject filament. A popup is displayed on the display to notify the user to load new filament and then confirm.

After this has been done the printer purges and continues the print correctly - however: the display keeps the popup open and does not return to the status screen:

[image]

### Steps to Reproduce

1. Enable filament runout sensor.
2. Start a print job from OctoPrint.
3. Snip the filament so a runout gets detected.
4. Load new filament and continue the print.

mosheas, 2021-03-04:
There is indeed a problem, attaching a link video describing the Display freezing phenomenon.
https://youtu.be/UMNVy43oLgw

Jyers, 2021-03-04:
Hmmmm, guess I forgot filament runout would also use the M600 popup. I'll get a fix for this pushed soon.

Jyers, 2021-03-04:
Alright, it's pushed but I don't have a runout sensor, so if one of you could give the fix a test, that would be great.

mosheas, 2021-03-04:
Yes, now it has been fixed working properly, thanks.

closed 2021-03-04 by Jyers