Jyers/Marlin #175: Display freezing after Filament Runout, in last Updated

mosheas, 2021-03-04:
After last Updated, during sensor filament runout  prosses the display get freeze, please see attached video show the phenomenon , thank you. 

https://youtu.be/UMNVy43oLgw

Schnabulation, 2021-03-04:
Duplicate of this: https://github.com/Jyers/Marlin/issues/174

mosheas, 2021-03-04:
this is a new problem happen on last Updated before few days

mosheas, 2021-03-04:
No, this is a new issue that appeared a few days ago in the last update, before that it worked fine.

Schnabulation, 2021-03-04:
If you would check my comment, you would see, that I created that issue **this morning.** So yes, this a new issue that arrived with the latest firmware.

mosheas, 2021-03-04:
OK, thanks. I will close it

mosheas, 2021-03-04:
Double issue

closed 2021-03-04 by mosheas

Schnabulation, 2021-03-04:
> OK, thanks. I will close it

Thanks. You can comment on my issue with your YouTube video - that helps a lot.