Jyers/Marlin #177: Item menu RUNOUT Distance issue

tititopher68, 2021-03-04:
The "Runout_distance" item in the Advanced menu does not appear regardless of what has been defined in Configuration.h (         
 //#define FILAMENT_RUNOUT_DISTANCE_MM or #define FILAMENT_RUNOUT_DISTANCE_MM).

Jyers, 2021-03-04:
Fixed with the latest commit.

closed 2021-03-04 by Jyers

tititopher68, 2021-03-04:
Which I have solved by comment //#define FILAMENT_MOTION_SENSOR in Configuration.h and modifying the file below as follows:

In Marlin/src/lcd/extui/lib/creality_dwin/creality_dwin.cpp :

@@ -1907,7 +1907,7 @@ void CrealityDWINClass::Menu_Item_Handler(uint8_t menu, uint8_t item, bool draw/
      #define ADVANCED_LOAD (ADVANCED_YOFFSET + ENABLED(ADVANCED_PAUSE_FEATURE))
      #define ADVANCED_UNLOAD (ADVANCED_LOAD + ENABLED(ADVANCED_PAUSE_FEATURE))
 -    #define ADVANCED_FILSENSORENABLED (ADVANCED_UNLOAD + ENABLED(FILAMENT_RUNOUT_SENSOR))
 +    #define ADVANCED_FILSENSORENABLED (ADVANCED_UNLOAD + ENABLED(HAS_FILAMENT_SENSOR))
 -   #define ADVANCED_FILSENSORDISTANCE (ADVANCED_FILSENSORENABLED + ENABLED(FILAMENT_RUNOUT_DISTANCE))
 +   #define ADVANCED_FILSENSORDISTANCE (ADVANCED_FILSENSORENABLED + ENABLED(HAS_FILAMENT_RUNOUT_DISTANCE))
      #define ADVANCED_TOTAL ADVANCED_FILSENSORDISTANCE

      switch (item) {
@@ -1974,6 +1974,7 @@ void CrealityDWINClass::Menu_Item_Handler(uint8_t menu, uint8_t item, bool draw/
            Draw_Menu(Advanced, ADVANCED_FILSENSORENABLED);
          }
          break;
    -   #if ENABLED(FILAMENT_RUNOUT_DISTANCE)
    +   #if ENABLED(HAS_FILAMENT_RUNOUT_DISTANCE
        case ADVANCED_FILSENSORDISTANCE:
          if (draw) {
            Draw_Menu_Item(row, ICON_MaxAccE, (char*)"Runout Distance");
@@ -1984,6 +1985,7 @@ void CrealityDWINClass::Menu_Item_Handler(uint8_t menu, uint8_t item, bool draw/
          }
          break;
        #endif
        #endif
      }
      break;
    case InfoMain:

Jyers, 2021-03-05:
Thanks for the tip on HAS_FILAMENT_RUNOUT_DISTANCE. That's a bit more elegant than my fix. I have implemented the new version