Jyers/Marlin #179: BL touch does not work.

Predator2911, 2021-03-05:
Hi,
[image]

used versions: E3V2-BLTouch-5x5-V4.bin; E3V2-BLTouch-5x5-HS-V4.bin; E3V2-BLTouch-3x3-V4.bin; E3V2-BLTouch-3x3-HS-V4.bin

With all the same, the Z-axis measurement will not start. takes it up but, it doesn't move down from the axes, only up and ends there with an error.

Plug in my BL touch.

[image]
[image]

Thank you in advance for the advice.

Schnabulation, 2021-03-05:
The pre-compiled firmware assume the BL Touch to be connected to the dedicated BL Touch port. As far as I see you are using a breakout board to plug it in - if you want to use that you need to compile your own firmware.

Predator2911, 2021-03-05:
And will I be able to do it in a way that I don't have to do my own firmware? If you can advise.

Schnabulation, 2021-03-05:
How does the plug of your BL Touch look like? There is a dedicated port on the mainboard - BL Touch needs to be plugged into there.

Predator2911, 2021-03-05:
I found one video now, could it be involved like this?
 [image]
[image]

Predator2911, 2021-03-05:
I did it and it works like this, thank you for the help. [image]

Pls close.

Predator2911, 2021-03-05:
I have a new problem, autoleveling goes off the board. [image]

t3chguy, 2021-03-05:
Set your probe offset in the Control > Advanced menu in the UI.

Predator2911, 2021-03-05:
Thank you, I got it. I'm sorry, I'm new to all this, so I don't understand. Thank you again.

Jyers, 2021-03-05:
No need to apologize, we all have to start somewhere :P

closed 2021-03-05 by Jyers