Jyers/Marlin #182: Manual Mesh Bed Leveling 3x3 doesn't work

mosheas, 2021-03-05:
BUG - Manual Mesh Bed Leveling doesn't work,the button Z position changes only number values, but does not actually move the z axis after pressing the button.

feelfree69, 2021-03-05:
here, it does work.

Jyers, 2021-03-05:
Is this using a precompiled version or your own build? If continue to see this issue, a very detailed description of how to recreate it or a video of the phenomenon may help to figure out the issue.

mosheas, 2021-03-05:
I took the latest update on the site, replaced the configuration file with manual mesh leveling menu 3x3  and built it. it was work fine until last Updated this morning.

Jyers, 2021-03-07:
Is this issue still occurring? I have not heard any other accounts of this happening.

Jyers, 2021-03-07:
I just took a look and immediately found the bug. Looks like one of the recent PR contributors made a mistake with a conditional and disabled some of the manual mesh code. It should be fixed with the latest commit, sorry for the confusion.

closed 2021-03-07 by Jyers