Jyers/Marlin #187: Capped steps per mm at 186

Axionkt, 2021-03-05:
### Description

At the moment there is a cap for extruder steps per mm set at 186, same as the ender 3 v2 stock firmware, however Marlin upstream has this problem fixed: 
[Marlin-firmware](https://github.com/MarlinFirmware/Marlin/pull/20324) 

No hurry as this is more a matter of convenience
Thanks for all your work.

### Steps to Reproduce

1. On the printer display go to Control
2. Motion > Extruder
3. Try to raise the value
4. It stops at 186

**Expected behavior:** [What you expect to happen]
Be able to set the extruder steps per mm to a value over 186
**Actual behavior:** [What actually happens]
The value is capped at 186.
#### Additional Information

Jyers, 2021-03-07:
I'll up the max to 999.9 as it is in the main branch :)

Jyers, 2021-03-07:
I did you one better and set it to 1000

closed 2021-03-07 by Jyers

Axionkt, 2021-03-08:
Great, thanks a lot!