Jyers/Marlin #190: E-step only to 186

Mr-Rog, 2021-03-06:
Hello 
I have buy a BondTech extrudor, in the Manual says i must setup the e-Steps to 415 but I check in the firmware gos only to steps/mm 186
is this possible to change this, will pay for the work no problem
thx for a short feedback

Schnabulation, 2021-03-06:
The issue is already known and will be fixed in the next release.

Consider donating - helps him a lot...

Mr-Rog, 2021-03-06:
Hi Schnabulation 
thx for your answer, how i can donate?
can I ask you a question I use for my Ender 3 V2 Bltouch this version
E3V2-BLTouch-5x5-HS-V4.bin
is this OK
also I mount a BondTech extrudor, I read I neet to setup in Cura  M92 E415 is this steps/mm xxx this setting?
sorry wor asking but I dont know person they use the same Firmware
thx and cheers

Schnabulation, 2021-03-06:
Oh hi there, fellow Swiss. :-)
If it‘s ok, I will keep stay with English so the others can understand us. But if you want, you can send me a PM and we can talk in German.

Regarding the e-steps: Best would be to set them in the firmware. For now it is only possible to set it to 186 through the menu - but if you self-compile the firmware you can go higher. I don‘t know the command to set it through Cura though.

And btw: on the main repo page of this firmware on the top right is a „Sponsor“ button. ;-)

tititopher68, 2021-03-06:
Just to modify line 2028 in \src\lcd\extui\lib\creality_dwin\creality_dwin.cpp :  

Modify_Value(planner.settings.axis_steps_per_mm[E_AXIS], 0, **default_steps[E_AXIS]*2**, 10);

by : 

Modify_Value(planner.settings.axis_steps_per_mm[E_AXIS], 0, **1000**, 10);

;-) !

Mr-Rog, 2021-03-06:
thy I will check, but when I donwload the code and check it for me line1895

Jyers, 2021-03-07:
Issue resolved with #187, will be available in the next release. If you want any more help compiling, I recommend opening a discussion thread :)

closed 2021-03-07 by Jyers