Jyers/Marlin #200: [BUG] Manual leveling issues "G420 Sn" instead of "M420 Sn"

Schnabulation, 2021-03-07:
### Description

In order to make the manuel leveling independent from the leveling mesh the board issues a command to disable the mesh. The correct command for this is `M420 S0` - however the printer recieves a `G420 S0`. This is not a known command.

After the leveling is complete the board restores the mesh. The correct command is `M420 S1` - however the printer recieves a `G420 S1`.

[image]

dafi87, 2021-03-07:
I noticed the same thing. Problem is 
https://github.com/Jyers/Marlin/blob/9c82eab3510bc60af153c068bc9ae5028c4c18f7/Marlin/src/lcd/extui/lib/creality_dwin/creality_dwin.cpp#L775
and
https://github.com/Jyers/Marlin/blob/9c82eab3510bc60af153c068bc9ae5028c4c18f7/Marlin/src/lcd/extui/lib/creality_dwin/creality_dwin.cpp#L936

Jyers, 2021-03-07:
Classic case of me programming way too late at night... Fixed with latest commit

closed 2021-03-07 by Jyers