Jyers/Marlin #205: Can't build v1.2.0 with TMC2208 Extruder (NOT standalone)

nullstalgia, 2021-03-08:
I can compile it with the supplied UBL config, and my Bilinear + Lin Advance, but combined gives an error concerning lines 463 and 464 in `creality_dwin.cpp`

[image]

I use [this guide](https://drive.google.com/file/d/15cUf10lMxW4NUHE9qVAlamXmULdrOJxz/view) to enable Linear Advance (I did the internal solder mod)

Jyers, 2021-03-08:
Just tried building with the provided configs and was not able to recreate the error. If those aren't actually the files that cause the error, a set of configs that I can use to recreate the issue would be ideal.

nullstalgia, 2021-03-08:
Ah crap, I think I uploaded the wrong one.

https://gist.github.com/nullstalgia/ab75ca7d2a2857a27fb900fb0012613d

This for sure gives me the error.

Jyers, 2021-03-08:
Unfortunately it still built perfectly on my end... Perhaps try a fresh pull and see if that helps.

Jyers, 2021-03-08:
The configs you just sent also still show the standalone extruder driver, don't know if that was intentional or not.

nullstalgia, 2021-03-08:
I HAVE THE WRONG FOLDER OPEN

Jyers, 2021-03-08:
That may be an issue... XD

nullstalgia, 2021-03-08:
The folder names all look the same, my bad. Here is a proper Config

https://gist.github.com/nullstalgia/007fa6f8476a1b5daaa4a93192d1d8f6

And as I was typing before I realized:

Hmm. I did that before I sent in the issue to make sure it wasn't a malformed pull or something... Did you follow the guide and edit `platformio.ini` and `pins_CREALITY_V4.h` ?

Jyers, 2021-03-08:
Okay, I have recreated the error. Let me see what I can do here.

Jyers, 2021-03-08:
Took me way to long to understand that error. Just needed a god damn type case... Latest commit fixes the issue.

closed 2021-03-08 by Jyers

nullstalgia, 2021-03-08:
Trust me, I almost went crazy looking at it too. Thanks for the help!

nullstalgia, 2021-03-08:
Hmm, the UBL is giving me some strangeness. I'm gonna do some testing with the precompiled binaries and other configuration settings.