Jyers/Marlin #208: UBL only probes ~65/100 and Menu Freezes after Auto Tilt

nullstalgia, 2021-03-08:
This occurs both on the precompiled firmware and self-compiled.

Serial Log:

```
Recv: Probing mesh point 59/100.
[...]
Recv: Probing mesh point 60/100.
[...]
Recv: Probing mesh point 61/100.
[...]
Recv: Probing mesh point 62/100.
[...]
Recv: Probing mesh point 63/100.
[...]
Recv: Probing mesh point 64/100.
[...]
Recv: Probing mesh point 65/100.
[...]
Recv: X:0.00 Y:0.00 Z:12.55 E:0.00 Count X:0 Y:0 Z:5020
Recv: echo:Bed Leveling ON
Recv: echo:Fade Height 10.00
Recv: ok
[...]
Recv: Done.
[...]
Recv: Tilting mesh point 1/4
Recv: 
[...]
Recv: Tilting mesh point 2/4
Recv: 
[...]
Recv: Tilting mesh point 3/4
Recv: 
[...]
Recv: Tilting mesh point 4/4 (at this point, the screen is frozen and won't respond)
Recv: 
[...]
Recv: X:249.00 Y:15.00 Z:12.55 E:0.00 Count X:19920 Y:1200 Z:5020
```

[image]

Also showing it stopping on # 65
[image]

The resulting mesh: 

[image]

gmergulhao, 2021-03-08:
I can confirm screen won't clear after mesh tilt. A G28 through Octoprint works. seems the display clear is missing

tititopher68, 2021-03-08:
Operation is normal. The number of points carried out depends on the position of the probe in relation to the nozzle (The head cannot go everywhere .... see doc Marlin).
In the code, a total Mesh is requested by G29 P1, but if it is incomplete, it is followed by another Gcode g29 P3.

As for the freeze after the auto tilt.
in \ src \ lcd \ extui \ lib \ creality_dwin \ creality_dwin.cpp :
add after line 2497: **Redraw_Menu ()**

either once modified:

line 2496 gcode.process_subcommands_now_P (PSTR (buf));
line 2497 planner.synchronize ();
line 2498 **Redraw_Menu ()**

;-)

Jyers, 2021-03-08:
I pushed a fix for the AutoTilt freeze. @tititopher68 is right about the fact that it will not be able to probe all points as the bed probe can not reach every point, it will infer the rest.