Jyers/Marlin #209: UBL Issues

darco415, 2021-03-08:
Where do I start? First off, unable to compile,,,,but Ill deal with that later. DL two different UBL bins, both have issues. 
Currrent one 10x10 HS v2,stuck on proboing,staying in one spot but probing points going up. The other thing is I CANNOT connect from Octopint anymore! I dont know whats going on here.  Tried different bins,cables and octoprint settings. I can only access menu ..uggh

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead use one of the following options:

- The Marlin Firmware forum at https://reprap.org/forum/list.php?415
- The MarlinFirmware Facebook Group at https://www.facebook.com/groups/1049718498464482/
- The MarlinFirmware Discord Server at https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the latest "bugfix" branch to see whether the issue is already addressed.

-->

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

nullstalgia, 2021-03-08:
I'm having the same issue, seems like a firmware bug.

mzielinskim, 2021-03-08:
Using the precompiled E3V2-UBL-BLTouch-10x10-v4.2.2.bin i can connect to octoprint.
Only while UBL-Probimmg it didnt connect...

nullstalgia, 2021-03-08:
That's common with any firmware, as far as I know. The printer can't
respond to Octoprint asking if it's alive during a probe. So it's best to
connect before/after.

gmergulhao, 2021-03-08:
I  could connect after disabling meatpack plugin. using 15x15 hs precompiled.

tititopher68, 2021-03-08:
Operation is normal. The number of points carried out depends on the position of the probe in relation to the nozzle (The head cannot go everywhere .... see doc Marlin).
In the code, a total Mesh is requested by G29 P1, but if it is incomplete, it is followed by another Gcode g29 P3.

As for the freeze after the auto tilt.
in \ src \ lcd \ extui \ lib \ creality_dwin \ creality_dwin.cpp :
add after line 2497: **Redraw_Menu ()**

either once modified:

line 2496 gcode.process_subcommands_now_P (PSTR (buf));
line 2497 planner.synchronize ();
line 2498 **Redraw_Menu ()**

;-)

Jyers, 2021-03-08:
I pushed a fix for the AutoTilt freeze. @tititopher68 is right about the fact that it will not be able to probe all points as the bed probe can not reach every point, it will infer the rest. I also just tested building both of the config templates with no errors, so I don't know why you would be having issues with that.

darco415, 2021-03-08:
So earlier I only disabled meatpack. Just now I totally uninstalled it and now using your pre-comp, I able to connect to Octoprint

Jyers, 2021-03-08:
Glad you got it working, btw I just release v1.2.1, which should fix the UBL issues you have encountered so far.

closed 2021-03-08 by Jyers

Friday13th87, 2021-04-02:
may i ask in this question what HS means in the firmwares?

Jyers, 2021-04-02:
HS stands for High Speed and refers to high speed probing with the BL Touch. Essentially the probe will not stow itself between probing moves, which is faster, but increases the possibility of it crashing if your bed is very tilted.

Friday13th87, 2021-04-02:
thanks very much - i saw it now in the wiki.
so if the bed is leveled good / very good i can run the HS version or does it lower the accuracy of the bl touch as well?

Jyers, 2021-04-03:
It shouldn't lower the accuracy of the bl touch, but I have not explicitly tested that fact, so take it as you will.