Jyers/Marlin #210: Lower the EXTRUDE_MINTEMP

funnybrum, 2021-03-08:
### Description
There are filaments that work on lower than the default 185C temperature. The Filamentum Timberfill for instance has recommended printing temperature starting at 140C. At 185C there is excessive stringing.

### Steps to Reproduce
Set the hotend temperature at 150C in the slicer and start printing.

**Expected behavior:** [What you expect to happen]
The printing process should be extruding filament.

**Actual behavior:** [What actually happens]
The printing process goes as expectef (X/Y/Z movements), but the E movement is blocked.

#### Additional Information
Should not be needed.

funnybrum, 2021-03-08:
And after a few seconds thinking about it - this probably should go against the official Marlin repo. I'll have an issue failed there if the repo owners think the same way.

t3chguy, 2021-03-08:
It is a safety feature to not damage the extruder.
You can use the Cold Extrude M302 to explicitly disable it from runtime
https://marlinfw.org/docs/gcode/M302.html

Schnabulation, 2021-03-08:
> And after a few seconds thinking about it - this probably should go against the official Marlin repo. I'll have an issue failed there if the repo owners think the same way.

I would also highly suggest to file this issue on the official Marlin repo. I'm a advocate to keep the Jyers-firmware as close to official Marlin as possible so he can get the firmware upstream. If we (or he in the end) changes too much settings to the baseline Marlin this makes it difficult.

Also: In the latest version you can set the "EXTRUDE_MINTEMP" directly on the display. It's under "Control" --> "Advanced". Or you compile your own firmware and set it there.

Jyers, 2021-03-08:
As @Schnabulation said, this is why I made the Extrude Mintemp modifiable from the advanced menu. It should be left at 180 as that is a safe baseline, but if you need it to be lower, simply change it from the menu or modify it in your config file.

closed 2021-03-08 by Jyers

funnybrum, 2021-03-08:
Did not know about the new feature in the latest version. Thanks!