Jyers/Marlin #222: V1,2,1 UBL no Mesh creation

mzielinskim, 2021-03-09:
### Description
Cannot read new Mesh.

### Steps to Reproduce

Instal new Firmware,Reset to defaults, set Probe Offset to x-45.0 y-7.0, UBL/Setting&Clear Current Mesh, Create New Mesh

It starts but then the Display says: Printer Kill reason: M112 Shutdown, Restart Reqired
And the BLTouch is blinking

When i shutdown octoprint prio to that -> Create New Mesh 
Display Says "Auto bed leveling, Please wait until done", Hotend goes up and left after that BLTouch is Blinking.

Hint: My Z-Axis was at about 20cm over the Bed before PowerOn - That caused the error - after manually going down to 1 cm over Bed level the Leveling started as it should.

tititopher68, 2021-03-09:
Hello,

As for this issue, , to secure certain actions (if for example, we deactivate the steppers then we manually move the x and / or y axes, the x, y, z positions can be shifted with a high risk of crashing on the Bed ...), but above all to improve the precision of the UBL and always according to Marlin, it is preferable to always start from the center of the Bed ...

Also I recommend adding an Auto Home G28.
(For me, with this in my version, it's OK, also with Octoprint)

#if HAS_BED_PROBE
            case UBL_GET_TILT:
              if (draw) {
                Draw_Menu_Item (row, ICON_Tilt, (char *) "Autotilt Current Mesh");
              }
              else {
                Popup_Handler (Level);
                **gcode.process_subcommands_now_P (PSTR ("G28"));**

                char buf [10];
                if (ubl_conf.tilt_grid> 1) {
                  sprintf (buf, "G29 J% i", ubl_conf.tilt_grid);
                }
                else {
                  sprintf (buf, "G29 J");
                }
                gcode.process_subcommands_now_P (PSTR (buf));
                planner.synchronize ();
                Redraw_Menu ();
              }
and 

case UBL_GET_MESH:
              if (draw) {
                Draw_Menu_Item(row, ICON_Mesh, (char*)"Create New Mesh");
              }
              else {
                Popup_Handler(Level);
                **gcode.process_subcommands_now_P(PSTR("G28"));**
                gcode.process_subcommands_now_P(PSTR("G29 P0\nG29 P1"));
                gcode.process_subcommands_now_P(PSTR("G29 P3\nG29 P3\nG29 P3\nG29 P3\nG29 P3\nG29 P3\nG29 P3\nG29 P3\nG29 P3\nG29 P3\nG29 P3\nG29 P3\nG29 P3\nG29 P3\nG29 P3\nM420 S1"));
                planner.synchronize();
                Redraw_Menu();
                Confirm_Handler("The mesh has NOT been saved");
              }
              break;

Note : I have also make correction in the code like : 
} elsif (....) {
by 
}
elsif (...) {

;-)

mzielinskim, 2021-03-09:
So this gies to the programmer that he shpould addd the G28 ? Because i wasnt aware oft thos and just used the display/firmware and thought it would do auto homing if needed.

Jyers, 2021-03-14:
This should be fixed with the latest commit, next release should be out soon with the most recent fixes.

closed 2021-03-14 by Jyers