Jyers/Marlin #223: Change Filament and Preheat Menu not working properly

AngleV, 2021-03-09:
### Description

Change Filanent from the prepare menu does not function properly

### Steps to Reproduce

From the main menu choose Prepare.
Then choose Change Filament -> Change Filament.
Popup informs you that temp is too low and asks if you want to open up the Preheat Menu.
Choose Yes.
Popup goes away and Preheat Hotend Menu appears. Temperature starts to rise automatically.
Now if you go down and choose Continue a new popup appears that informs you to wait until temperature is correct.
As soon as the temperature reaches the predefined value in this case 180 Celcius the popup dissappears and then you get a blank screen.

tititopher68, 2021-03-09:
Hello,

As for this issue, possible correction by : 

in \ src \ lcd \ extui \ lib \ creality_dwin \ creality_dwin.cpp :
add after line 3190 and line 3211: **Redraw_Menu ()**

Note : I have also make correction in the code like : 
} elsif (....) {
by 
}
elsif (...) {

;-)

nullstalgia, 2021-03-10:
@tititopher68 That kind of change seems like it would be best put in a pull request. :)

Legsmaniac, 2021-03-10:
I applied the above fix out of curiosity and compiled.
At first I thought it didn't work.
I select Change Filament.
A message says Nozzle is too Cold, Open Preheat Menu?
I select Confirm.
Presented with the preheat menu and automatically preheats to 180 and stays there.
I select Cancel.
Takes me back to the Change Filament Menu and starts cooling down.

Now, I didn't realise that you need to scroll down to select Continue which was off the bottom of the screen.
But this time, while it was preheating to 180, I selected PETG because I need a higher temp than 180 for the PLA+ that was currently in there..  If the temp is not hot enough, I found the extruder wheels slip on the filament.
So this time I scrolled down and selected Continue and sure enough, it went back to  the usual menu of extruding filament, please wait but the temp also immediately changed back to 180 despite selecting PETG, 240C
Fortunately, it was still hot enough to retract the filament but when loading the filament back in, it couldn't do the usual extrude of a portion of filament at the end as 180 was too cold for PLA+ resulting in the extruder wheels slipping and grinding on the filament and barely anything extruding from the nozzle.
Then it stays at 180C unless I select Cooldown?  Maybe it times out and turns off after a while, I don't know but it's not turned off the whole time it took me to write this up and I'm a slow typer!  (Two finger hunt-and-peck method!)

Why the need to go to the Preheat menu and still only automatically set 180 even though I selected a higher temp?
I prefered the old way where you just got a menu pop up saying something like Preheating nozzle, please wait and then proceeded to retract the filament. (Although it would probably be better if it preheated to 200 rather than 180?)

Hope this helps.

Legsmaniac, 2021-03-10:
PS
Just discovered that it also leaves a 1kB file on the SD Card called PLR dated 1/1/2000 01:00  ?!?!

talexop, 2021-03-10:
I search also for this issue and it looks like a bug to me..
My proposal is to change **m**enu to **M**enu at line 350 of \src\lcd\extui\lib\creality_dwin\creality_dwin.cpp.
It seems that might be the issue of not storing the last_menu selections.
I have not tried it yet.

inline void CrealityDWINClass::Draw_Menu(uint8_t menu, uint8_t select/*=0*/, uint8_t scroll/*=0*/) {
  if (process==**Menu** && active_menu!=menu) {
    last_selection = selection;
    last_menu = active_menu;
  }

tititopher68, 2021-03-10:
The proposal I made works and seems to correspond to what was expected. In any case, this is the one I use.

You can test my file version by copying it to \ src \ lcd \ extui \ lib \ creality_dwin \
The link: https://e.pcloud.link/publink/show?code=XZmB3kZ84GMeXm31lbBaf2mWBB1ayRoweSk

;-)

talexop, 2021-03-10:
I agree with you that it works.. but maybe the fixture I propose solves also other issues.. As I said I have not tried it yet. Maybe your fix is needed also!

talexop, 2021-03-10:
I have tested it! For sure it was a bug so menu has to be changed with Menu. No blank screen anymore. But when I preheat the extruder after the popup to wait it returns back to menu with the preheat option... So I think we should return with "Draw_Menu(Prepare, PREPARE_CHANGEFIL);" As @tititopher68 suggested

nullstalgia, 2021-03-10:
> PS
> Just discovered that it also leaves a 1kB file on the SD Card called PLR dated 1/1/2000 01:00  ?!?!

PLR is the Power Loss Recovery. Every set amount of Z steps, it saves where it is in the print onto the SD card, so if power is lost, it can attempt to resume at around the same spot.

Legsmaniac, 2021-03-10:
> PLR is the Power Loss Recovery.

Ah, OK, thanks.  Just never seen it before.

tititopher68, 2021-03-12:
Modications now are OK,
Microstep up and down issue -> OK
Popup PidTune and other PidBad...issue ->OK
Change filament and Prehead issue -> OK

Download link :
https://e.pcloud.link/publink/show?code=XZEc3kZCGY4gq3kdU4XsxQ5vGYHhYf6RsAk

In this file :

Configuration's files for ULB 5x5 Noprobe (motherboard v4.2.2 et v4.2.7)

File modified : creality_dwin.cpp (to be copied in folder "/src/lcd/extui/lib/creality_dwin/"
DWIN_SET (to flash the LCD)

;-)

Jyers, 2021-03-14:
This should be fixed with the latest commit, next release should be out soon with the most recent fixes.

closed 2021-03-14 by Jyers