Jyers/Marlin #224: Filament Runout and Advanced Menu settings

Legsmaniac, 2021-03-09:
Finally got around to downloading and trying out the latest Extensible_UI firmware.

Loving the new settings in the Advanced Menu!  👍
I see this also includes option to turn on/off the Filament Sensor but what about whether users have sensors either normally HIGH or normally LOW as defined in `Configuration.h` with the `define FIL_RUNOUT_STATE` setting?  Could there not be an option for this too?
Also what about those who are using rotary runout sensors such as the (apparently popular) BTT Runout Sensor?  Another setting for this?  Otherwise users have to uncomment `define FILAMENT_MOTION_SENSOR` in `Configuration.h` and compile their own firmware.
I think the inclusion of options in Advanced is great for new users who have no idea how to compile their own firmware so the above missing features could well be a good addition also, I feel.

Finally, a totally unrelated question if I may - why, in your config files is the E3V2 bed size set to 230x230 when it should be 220x220?  Is this intentional?  Does this give us a slightly bigger print size?

t3chguy, 2021-03-09:
`FIL_RUNOUT_STATE` can only be defined at compile time in Marlin. It cannot be changed at runtime. https://github.com/Jyers/Marlin/blob/713de872ce055bd10bd59b4f811136b0ec0cfbae/Marlin/src/module/endstops.cpp#L486

Same story for `FILAMENT_MOTION_SENSOR`.
You should make an issue with mainline Marlin to support changing those at runtime.

> why, in your config files is the E3V2 bed size set to 230x230 when it should be 220x220? Is this intentional? 

The Ender 3 /v2 Bed is actually physically 235x235 and the machine can move a little bit more than that yet, 220 is a very conservative limit but 230x230 if centered by endstops properly gives you bigger print size whilst still fitting in the physical volume.

Legsmaniac, 2021-03-09:
Thanks for the reply, makes sense.  Shame about the settings.

Legsmaniac, 2021-08-28:
Reopening this issue because I'm guessing this can be done now?

Recently I tried this firmware by mriscoc which has Filament Runout State in the Advanced Menu settings on the LCD.

It would be great if this feature could be added to Jyers because it simply cannot be beat.
I noticed some other great features in mrisocs variant such as Selectable mesh points from 2x2 to 7x7, PID popup with temperature plot, Filament run-out sensor with active state setup (LOW/HIGH), Background Colour Change.

And before anyone asks why I don't go use mrisoc's firmware then - despite some cool features, I'm not convinced it's as good as Jyers, plus I get compile errors with his, I would much prefer to stick with Jyers, with or without these features.  Just thought it would be nice if some, the filament runout state at least, could be included.

Thanks for reading and thanks for a truly great firmware.

eaytac, 2021-09-29:
I have this problem, is it possible to add "Filament Runout State" to menu?
I'll also appreciate if this option added. Thank you so much for this firmware.