Jyers/Marlin #229: UBL levelling microstep up and microstep down feature not working

alibarut, 2021-03-09:
Hi Jyers,

I have ender 3 v2. I havent BLTOUCH. 

I Flashed E3V2-UBL-NoProbe-5x5-v4.2.2.bin. 

UBL levelling microstep up and microstep down feature not working. Only work change the z-offset value.

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tititopher68, 2021-03-09:
Hello,

Do you active : Go to Mesh Z Value ? Without that,  microstep up or down not working...

Legsmaniac, 2021-03-10:
I concur, it's not working.
And yes, I selected Go to Mesh Z Value.
Also flashed E3V2-UBL-NoProbe-3x3-v4.2.2.bin to test just in case I'd done something wrong in my self-compiled one originally.
Just the same - not working. 
What is also weird is if I use the Point Z Offset and set to a value where it touches the paper on the bed and then go back to the same point later, the nozzle is way off the bed again even though the Point Z Offset still shows the value I last set, resulting in my having to adjust Point Z Offset again.  I'm sure this will result in an incorrect Mesh.  
Also, I have a previously perfectly leveled bed yet the nozzle is way up high off the bed and I can only set the Z Offset to a max of -1.00, it won't go any lower so I couldn't set the Mesh.  Really strange.
Unless I'm missing something or doing something else wrong?

tititopher68, 2021-03-10:
Very étrange...
Did you save the z-offset (Save button in the menu)?

Legsmaniac, 2021-03-10:
> Did you save the z-offset (Save button in the menu)?

You mean the Save that appears after the final "Next Point" in the menu, where the Next Point changes to Save?  Then yes, of course.

Legsmaniac, 2021-03-10:
> No, in the Z-Offset menu, not in the UBL menu

Oh, ok.
But you'll have to wait 10 hours now, sorry.
Machine is now doing a print.
Unless someone else has a go?

tititopher68, 2021-03-10:
You can test my file version by copying it to \ src \ lcd \ extui \ lib \ creality_dwin \
The link: https://e.pcloud.link/publink/show?code=XZmB3kZ84GMeXm31lbBaf2mWBB1ayRoweSk
and recompile all....

Legsmaniac, 2021-03-10:
Just tried this.  Compile failed.

tititopher68, 2021-03-11:
Sorry, but i have no problem...i maked a lot of version for BLT, BLT+ULB, UBL noprobe, Mesh manuel, etc...no problem, no failed....So i don't understand...
Can you link your Configuration.h ans Configuration_adv.h ?

tititopher68, 2021-03-11:
I compile with my version 2 firmwares : 

E3V2-UBL-5x5-Noprobe-422.bin (for v4.2.2 motherboard)
E3V2-UBL-5x5-Noprobe-422.bin (for v4.2.7 motherboard)

You have to flash the LCD before with the last DWIN_SET !

Download link : https://e.pcloud.link/publink/show?code=XZEc3kZCGY4gq3kdU4XsxQ5vGYHhYf6RsAk
(in this file, the 2 firmwares, configuration's files for v422 and v427, DWIN_SET)

;-)

Legsmaniac, 2021-03-11:
It's OK, panic over!  Sorry, my bad.  I realised this morning that I had dropped the file in the wrong place.
So I corrected this and compiled successfully.

Went to Change Filament, sure enough it said the nozzle was too cold, did I want open Preheat Menu.  Yes.
I then selected Continue, Heating, Please wait until done, just like it used to.  Good so far.
It reached the set temperature then.......
Then the please wait message disappears and it just goes back to the Prepare menu.  Temp stays hot.  No filament is ejected.
I have to select Change Filament AGAIN! Then it works, ejects the filament with the usual wait and prompt messages.

So it does work, but not quite right.

Legsmaniac, 2021-03-11:
Further update - 
Again, my bad, realised I've reported the Filament Change but it's in the wrong section!  Doh!
Hey, it's early, ain't had my coffee yet!!  😜

By the way, I had already flashed the DWIN.  
Off to recompile or try one of the pre-compiled and the leveling, back in a bit.
Sorry again.

Legsmaniac, 2021-03-11:
OK, I downloaded your 5x5 no-probe for my 4.2.7 board and flashed.
Microstepping still doesn't do anything.
And yes, I checked Go to Mesh Z Value.  I also saved the Z Offset as suggested above and tried again, still microstepping doesn't move the nozzle.

Legsmaniac, 2021-03-11:
I noticed there's a Live Adjustment check box in the Z-Offset menu which has to be selected for the nozzle to move when adjusting in this menu.  Sadly, leaving this checked then going to Level and trying microstepping doesn't work either.  Back to the Z-Offset menu shows the Live Adjustment box becomes unchecked again.,

Should there perhaps be a Live Adjustment checkbox in the Level menu?  Or is this why microstepping isn't working because there's no live adjustment?
Just a thought.

tititopher68, 2021-03-11:
Sorry, I just retested and you are right. only the adjustment of the level of each point is done correctly by modifying the value of the Z-Offset (Menu UBL!).
I understood where the problem is, I am in the process of making the modifications ...

tititopher68, 2021-03-11:
Modications OK,

Download link : 
https://e.pcloud.link/publink/show?code=XZEc3kZCGY4gq3kdU4XsxQ5vGYHhYf6RsAk

In this file : 
Configuration's files for ULB 5x5 Noprobe (motherboard v4.2.2 et v4.2.7)
2 firmwares for UBL 5x5 Noprobe pre-compiled
File modified : creality_dwin.cpp (to be copied in folder "/src/lcd/extui/lib/creality_dwin//"
DWIN_SET (to flash the LCD)

;-)

Legsmaniac, 2021-03-11:
What happened to the last posts with the new modifications?
Seems to have vanished.  Deleted? I downloaded and tested, seems it's working now.
However, I discovered a new problem.
When leveling points 2,0 and 2,4 (5x5 menu) or points 1,0 and 1,2 (3x3 menu) the nozzle sits on the glass retaining clip which clips my glass sheet to the bed.  Obviously, the points could be different for those who use the bulldog clips but shouldn't be a problem at all for those who use the magnetic sheet flexible beds.  For those who do use clips, it's a problem and so obviously the nozzle is too close to the edge.  BLT was never this close to the edge.
Not really a good idea to keep moving clips when leveling.

tititopher68, 2021-03-12:
Hélas, non la modification ne fonctionne pas correctement. Elle affecte en l'état, le Z-Offset quand on a un BLTouch....donc pas bon.
Je suis en train de tester autre chose....

Jyers, 2021-03-12:
I'm going to fix this soon, the way the UBL movement was coded is somewhat strange and needs some TLC on my part.

tititopher68, 2021-03-12:
Modications now are OK,
Microstep up and down issue -> OK
Popup PidTune and other PidBad...issue ->OK
Change filament and Prehead issue -> OK

Download link :
https://e.pcloud.link/publink/show?code=XZEc3kZCGY4gq3kdU4XsxQ5vGYHhYf6RsAk

In this file :

Configuration's files for ULB 5x5 Noprobe (motherboard v4.2.2 et v4.2.7)

File modified : creality_dwin.cpp (to be copied in folder "/src/lcd/extui/lib/creality_dwin/"
DWIN_SET (to flash the LCD)

;-)

Jyers, 2021-03-13:
@tititopher68 Thanks for your work, but I would highly advise if you want to help by making fixes like theses to put them in a pull request, that way I can easily checkout changes and better document what is happening with the firmware. If you have not made a PR before and have any questions about how to I'm happy to help. Thanks again for all your work.

Jyers, 2021-03-14:
This should be fixed with the latest commit, next release should be out soon with the most recent fixes.

closed 2021-03-14 by Jyers