Jyers/Marlin #230: UBL mesh is not deactivated when manual leveling

xantrk, 2021-03-09:
### Description
I am using manual UBL. I have compiled my own based on example config.
Previously, there was commit #9ff1c1e844d17f4ac1dc10a4d63f96873f8d1b6e which disables bilinear mesh when manual leveling. I believe this should also be the case for UBL also. But it is not. Tried multiple times. UBL toggle in leveling menu, affects manual leveling menu.

### Steps to Reproduce

1. Manual leveling via manual levle menu
2. UBL level via leveling menu  (manual probing)
3. Go to manual level menu again
4. The bed appear tilted at probing points
5. Go to leveling menu to disable UBL
6. Go to manual level menu
7. The bed is perfectly level
8. Go to leveling menu enable UBL
9. The bed appears tilted at probing points

**Expected behavior:** 
Manual level should disable UBL mesh after it homes in order to be reproduble.

**Actual behavior:** 
UBL mesh sticks.

Jyers, 2021-03-09:
That is strange, as far as I can tell M420 S0 should be fully capable of disabling the UBL mesh, but perhaps G29 D is required instead. I'll do some testing.

tititopher68, 2021-03-09:
Hello Jacob,

I think it's just a probem with @xantrk'file Configuration.h...
I have just tested and no issue...

xantrk, 2021-03-11:
Just to give another update, i realized today that my z stop is phantom triggering i.e activates without the click. I tried homind and manual leveling like 10 times, and 2 times were z stop was triggerred with no clicking. Replaced it with Y stop. It appears my leveling is more consistent now. Meanwhile I am closing the issue, as it appears I was the only one and my limit switch was causing inconsistencies.

closed 2021-03-11 by xantrk

Jyers, 2021-03-12:
Glad to hear its working better now! Feel free to update this if you notice the issue again, as it could very much be an issue of how UBL handles getting disabled, but hopefully it ends up just being the hardware issue.