Jyers/Marlin #238: [FR] When printing from OctoPrint show the actual filename instead of "Host Print"

Schnabulation, 2021-03-11:
### Description

One of the biggest advantages of this firmware is when printing from a host like OctoPrint the DWIN display switches to the print screen and displays the same informations as when printing from SD card.
...except for one thing: on the top it displays "Host Print" instead of the actual filename:

[image]

Would it be feasible to get the filename from OctoPrint and then display this? Maybe have "(Printing from OctoPrint)" on a second line or something...

I could not find any information regarding the functions needed to query the filename from OctoPrint and I believe this would be necessary to send the string to the display. Maybe someone else knows this - or else this can be closed. It's just an idea...

jjfs127, 2021-03-11:
Asked this question as well in a previous issue. I suspect that the reason this may not already be implemented is possibly due to the fact the maybe only the dos filename is available (8.3 w/extension). But I could be wrong.

Jyers, 2021-03-12:
I haven't looked into this in a while, as it is easy enough to send the filename as part of your M117 information, but now that I think of it, I could implement a format of some kind which told the display to use the M117 as the filename above. It would require some way of sending the filename on connect, which might require coding a plugin or something of the like, but it would be possible.

Majestry, 2021-03-12:
As workaround, you can use Cura with Post-processing plugin.
There is an option Display filename and layer on LCD - it will add next commands to Gcode file for each line
`M117 Layer 1 of 44 {file name}`
. But if you have DsiplayLayerProgress and it's configured to send M117 commands - then ,most probably, it will earlier or later overwrite the M117 from gcode.

Schnabulation, 2021-03-12:
Thank you, I appreciate your workaround. However: I already have DisplayLayerProgress installed and as you can see it displays flawlessly all M117 commands. But M117 sets the LCD message which gets written into the dedicated text bar in the middle of the screen. I would like to have "Host Print" replaced with the filename.

But this is just a "nice to have" and nothing deal breaking as I can just check the filename in OctoPrint or my iOS app. It would certainly make the display look a bit cleaner.

Jyers, 2021-03-14:
This has been implemented with the latest commit. All you need to do it add "M117 <F>{{ event.name }}" after M75 in your octoprint "Before print job starts" gcode script.

closed 2021-03-14 by Jyers

Schnabulation, 2021-03-15:
Wow, this is great new - thank you very much for this!