Jyers/Marlin #242: [Issue] PID autotune notification is not closed

Majestry, 2021-03-11:
### Description

When PID autotune is started from the terminal, the pop-up appeares on screen when it's finidhed, but click on "Continue" makes nothing

### Steps to Reproduce

1. Start autotune from serial port using the command M303 E-1 S55 U1
2. When autotune is finished, popup window is shown on screen
3. Click in "Confirm"

**Expected behavior:** Pop-up is closed

**Actual behavior:** Pop-up isn;t closed

#### Additional Information

* Flashed firmware is v1.2.1 E3V2-UBL-BLTouch-10x10-HS-v4.2.7.bin
* Photo of pop-up 
[image]

tititopher68, 2021-03-12:
Modications now are OK,
Microstep up and down issue -> OK
Popup PidTune and other PidBad...issue ->OK
Change filament and Prehead issue -> OK

Download link :
https://e.pcloud.link/publink/show?code=XZEc3kZCGY4gq3kdU4XsxQ5vGYHhYf6RsAk

In this file :

Configuration's files for ULB 5x5 Noprobe (motherboard v4.2.2 et v4.2.7)

File modified : creality_dwin.cpp (to be copied in folder "/src/lcd/extui/lib/creality_dwin/"
DWIN_SET (to flash the LCD)

;-)

Jyers, 2021-03-14:
This should be fixed with the latest commit, next release should be out soon with the most recent fixes.

closed 2021-03-14 by Jyers