Jyers/Marlin #243: Latest firmware blocking Serial1 UART?

Legsmaniac, 2021-03-11:
### Description

I use ESP3D which is Arduino connected to the mainboard via the Serial1 UART port.
Never had a problem connecting before until I installed the latest new firmware.

### Steps to Reproduce

1. Install latest Extensible UI
2. Connect to ESP3D
3. Try sending commands or check Marlin settings page

**Expected behavior:** [What you expect to happen]

Commands send or Marlin Settings page loads connected.

**Actual behavior:** [What actually happens]

Marlin Settings page won't connect hence cannot send commands.

Jyers, 2021-03-13:
The only recent change I have made which could effect this could perhaps be BINARY_FILE_TRANSFER. If anyone is facing this issue and is compiling their own FW, try disabling this setting and let me know if it fixes the issue.

Turismo05, 2021-03-14:
I flashed E3V2-UBL-NoProbe-3x3-v4.2.2.bin and mine lost connection with PC also. I couldn't get the UBL to work so flahed back to the previous version and it connects again.

Jyers, 2021-03-14:
@Turismo05 Could you give this bin a try for me? I unfortunately don't have any way to recreate this issue.
[E3V2-UBL-NoProbe-3x3-v4.2.2-Testing.zip](https://github.com/Jyers/Marlin/files/6137197/E3V2-UBL-NoProbe-3x3-v4.2.2-Testing.zip)

Turismo05, 2021-03-14:
Hi, Yes this works. I can connect to pc and send commands.

Legsmaniac, 2021-03-14:
> ...... could perhaps be BINARY_FILE_TRANSFER

> Yes this works.

That's good to know, thanks.  I haven't had time to get around to trying it.  There's been a sad young death in the family so printing etc. has taken a back seat for the moment.

Turismo05, 2021-03-14:
So sorry to hear that, thoughts with you and your family.

Jyers, 2021-03-15:
Best to you and your family in these hard times, so sorry for your loss.

Jyers, 2021-03-15:
The commit "Meatpack Support" disables binary file transfer in favor of meatpack support so this should now be fixed and will be in the next release.

closed 2021-03-15 by Jyers